Koenkk/Z-Stack-firmware #332: CC2652P appear to be bricked after flashing

enboig, 2021-12-01:
I may have bricked my device. After updating zigbee2mqtt it stopped working, so I tried to identify the device
```https://www.ebay.es/itm/224578897533```
and flashed
```sudo ./cc2538-bsl.py -e -w -v -p /dev/tty  CC1352P2_CC2652P_other_coordinator_20210708.hex```

Now when I plug it while pushing buttons, leds aren't blinking

It may not be fully bricked, as lsusb returns:
```Bus 001 Device 064: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter```

But I don't know how to fix it, or which is the correct firmware.

Anybody can help me?

thanks

enboig, 2021-12-01:
It appear it was a "led" bug, I have been able to flash:
https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator

enboig, 2021-12-01:
The device appear to work again with this firmware, but I would like to know which one of this repository should I have used.

closed 2021-12-01 by enboig

RealKoenisch, 2022-04-21:
I recognise the same behaviour with the actual version - Zig Star V4 CC2652P - in Home Assistant despite of the parameter led_mode: on on led activities. In the former version no problem. Behaviour as descriped.

Please reopen

RealKoenisch, 2022-04-29:
I think it is better to open a new issue.