Koenkk/Z-Stack-firmware #333: C2531 coordinator keeps on crashing

tb205gti, 2021-12-02:
My coordinator has started crashing quite badly, to the point where it cannto run for more than a few hours to minutes. Once it crashed I have to re-flash the fw to it again to get it running again. Once it dies the led is very faint red.

Coordinator fw: 20211115, but it also happened on the previous fw.
Zigbee2mqtt is 1.22.1
I have a total of 35 zigbee devices, mostly Xiaomi temperature/humidity sensors and door sensors - but also Ikea switches. 6 of them are routers within reach of the coordinator.
Running Zigbee2mqtt on Ubuntu LTS 18.04, not running in docker container. Coordinator is connected via USB extension cord, and with additional antenna.

How can I start the debugging to see what is happening?

hacker-h, 2021-12-03:
have you considered that your hardware is defective e.g. memory?

tb205gti, 2021-12-07:
That is a possibility yes, but unless I can get to debug it I cannot be sure about it.

I just noticed that I get this is the logs:

(node:3299) UnhandledPromiseRejectionWarning: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt_1.22.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
(node:3299) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:3299) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Which indicates an issue with zigbee2mqtt and not the z-stack firmware.

Koenkk, 2021-12-07:
Check the points from https://www.zigbee2mqtt.io/guide/faq/#zigbee2mqtt-crashes-after-some-time (consider upgrading to a more powerful adapter)

tb205gti, 2021-12-08:
I have ordered a Electrolama, but it takes some time before it gets here, I'd rather not have the entire zigbee network crash all the time until then. I'll take a look at the link

closed 2022-01-16 by github-actions[bot]