Koenkk/Z-Stack-firmware #335: Errors while compiling CC2531: Undefined external

imjoel, 2021-12-06:
**Followed the compile.md guide for CC2531, (almost) every single step:**

1. Copied firmware.patch to Z-Stack Home 1.2.2a.44539
2. Opened Git Bash, go to Z-Stack Home 1.2.2a.44539 and apply the patch using git apply firmware.patch --ignore-space-change.
3. Opened Z-Stack Home 1.2.2a.44539\Projects\zstack\ZNP\CC253x\znp.eww with IAR Embedded workbench for 8051.
4. Converted version for both
5. Selected CC2531 as target under the workspace window tab: Overview | CC2530 | >>**CC2531** <<
6. Selected **- ProdHex** 
7. Right-click on your CC2531 - ProdHex and clicked options. Went to C/C++ compiler -> preprocessor. Removed all Defined symbols and added `FIRMWARE_CC2531` to Defined symbols:\
8. Clicked OK, Right-clicked on target and clicked Rebuild all.

I got 2 errors while compiling:
`Error[e46]: Undefined external "?B" referred in BindingTable ( C:\Texas Instruments\Z-Stack Home 1.2.2a.44539\Projects\zstack\ZNP\CC253x\CC2531-ProdHex\Obj\BindingTable.r51 ) 
Error[e46]: Undefined external "?IE" referred in ?ALLOC_XSTACK8 ( C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\8051\LIB\CLIB\cl-pli-blxd-1e16x01.r51 ) 
Error while running Linker 
 
Total number of errors: 2 
Total number of warnings: 3 
`

imjoel, 2021-12-07:
Searched long and hard. Apparently, the latest version of IAR (10.40.1) causes linking errors. Reverting to 10.10.1 worked for me with no issues

closed 2021-12-07 by imjoel

RaNo99, 2022-05-10:
I'm getting exactly the same issue. Fresh install of IAR 10.40.1 and Z-Stack.

RaNo99, 2022-05-10:
I found the solution. Go to project options, and in the "Linker" section identify the *.xcl file that is in use listed there (in my case it was `$PROJ_DIR$\..\..\..\Tools\CC2530DB\f8w2530.xcl`). Add the following lines to the end of it:
```
//
//  Device specific symbol definitions
//  ==================================
-D?B=F0                       // B register location
-D?IE=A8                      // Interrupt Enable register location
-D?IP=A9                      // Interrupt Priority register location
```
(these lines can be found in `C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\8051\config\devices\Texas Instruments\lnk51ew_cc2530F256.xcl`)

Clean and rebuild the project.

ThevinduSam, 2023-08-23:
@RaNo99 
how do you open that .xcl file inorder to edit it?

baylf2000, 2023-09-14:
> I found the solution. Go to project options, and in the "Linker" section identify the *.xcl file that is in use listed there (in my case it was `$PROJ_DIR$\..\..\..\Tools\CC2530DB\f8w2530.xcl`). Add [...]

Thank you! That worked for me!