Koenkk/Z-Stack-firmware #337: LP-CC2652RB - Failed to connect to the adapter

renatomotorline, 2021-12-10:
Hi,

I bought a [LP-CC2652RB](https://www.ti.com/tool/LP-CC2652RB) and I flash the firmware and try this code:

package.json
```
{
  "name": "zigbee-gateway",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "start": "node gateway.js"
  },
  "keywords": [],
  "author": "",
  "license": "ISC",
  "dependencies": {
    "zigbee-herdsman": "^0.13.179"
  }
}
```
gateway.js
```
const { Controller } = require('zigbee-herdsman')

const SERIAL = '/dev/ttyACM0'
const DB = './devices.db'

const coordinator = new Controller({
  serialPort: { path: SERIAL },
  databasePath: DB
})

coordinator.on('message', async (msg) => {
  console.log(msg)
})

coordinator
  .start((err) => {
    if (err) {
      console.error(err)
      process.exit(1)
    }
  })
  .then(async () => {
    console.log('started with device', SERIAL)
    coordinator.permitJoin(600, (err) => {
      if (err) {
        console.error(err)
      }
    })
  })
```

```
$ ls /dev/ttyACM*
/dev/ttyACM0  /dev/ttyACM1
```

But get this error:
```
node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:83
                    throw new Error(`Failed to connect to the adapter (${e})`);
                          ^

Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```

Firmware tested:
[CC2652RB_coordinator_20210708.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_coordinator_20210708.zip)
[CC2652RB_router_20210128.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC2652RB_router_20210128.zip)
Also tried the same firmware used in this issue [#284](https://github.com/Koenkk/Z-Stack-firmware/issues/284)

I use Windows to flash following this [instructions](https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_via_uniflash.html)
I use Fedora 25 to run nodejs
Any idea what I'm doing wrong?

renatomotorline, 2021-12-13:
Tried on Windows 10 and it works fine. It's a problem with Fedora 25. Any ideas?

jnxxx, 2022-01-12:
Try to switch from `/dev/ttyACM0` to `/dev/ttyUSB0`.
I changed from CC2531 to CC2652RB, and experienced a similar error when starting zigbee2mqtt until I changed that.

Or use `ls -l /dev/serial/by-id` to list your devices.

Also have a look at slaesh's recommendation: https://slae.sh/projects/cc2652/#homeassistant-hassio

closed 2022-02-20 by github-actions[bot]