Koenkk/Z-Stack-firmware #338: Adding devices via Coordinator not working with newer versions of Z-Stack_3.x.0 firmware than 20210120

iedex, 2021-12-10:
Hi,
I used for few weeks an Sonoff Zigbee 3.0 USB Dongle Plus with the preflashed firmware (20210120). Worked ok, but low LQI. So, I compiled the latest version from develop branch (20211207) with 20db TX power. The LQI improved (2x - 3x) but I couldn add devices using "Add Devices via this device" from HA. After that, I flashed latest firmware from master branch (20210708), pre-compiled, available here on github. Same issue.
Flashed 20210120 version (the one that the dongle had in the first place) and it works again (adding devices via coordinator/ this device).
Maybe something to do with `Fix joining not working when joining is only permitted on specific router` in 20210708 version?

iedex, 2021-12-11:
https://github.com/Koenkk/Z-Stack-firmware/issues/316#issuecomment-960886805

sphonk, 2021-12-12:
HI,

facing exactly the same issue, even worst because I cannot find version (20210120) so i have to go back to my old CC2531 it detects the devices, and then I move back to the Sonoff Zigbee 3.0 USB Dongle Plus, its a work around but kind of annoying

iedex, 2021-12-12:
@sphonk https://github.com/Koenkk/Z-Stack-firmware/blob/5adbe76236167e17ee24588e70ba35850f7061ab/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_20210120.zip

Lefuneste83, 2021-12-15:
Same issue here. Like many other, we are having issues with this dongle with the latest version of the FW. I have rolled back to my Modkam V3 based on cc2538 + cc2592. The power is not as strong as with the Sonoff + latest FW, but I can pair all my devices painlessly.

I built the Modkam myself and tried all sort of FW on it and most of them had the same issue than with the Sonoff latest FW. I recently built the Alternate version (with slightly different daughter PCB) of the Modkam V3 and it worked much better straight away, than previous iteration based on the exact same chipset.

On the original design depending of the FW version I tried, I encountered the exact same issue as with the Sonoff latest FW. Some devices would pair totally fine (Xiaomi doors, movement, Blitwolf plugs) while other would hardly pair (Tuya 3 buttons remote or Tuya plugs). Depending of the FW I used they would work or not. With the Modkam V3 Alternate version, although based on the same chips in theory, the recommended FW works perfectly and I didn't have to tinker at all.

I am waiting for my PCBs and parts for a Modkam V4 build and I'll see if it works fine.

I believe that there is a certain level of discrepancy between some FW versions and some hardware, making certain types of ZB devices to fail to pair at all. In my case they would either very hardly pair (but I could see the failed handshake), or not be detected at all (totally empty prompt), while Xiaomi devices would pair fine in all cases. Obviously this is all very frustrating.

On top of that, it seems, from a very basic perspective, that some FW are more adapted to certain flavors of the ZB protocols, and not all ZB devices manufacturers stick to the ZB specifications in the same way, which could be the cause of the FW misbehavior. I have a much higher success rate of pairing with Xiaomi and Blitzwolf devices than with any other brand. Tuya devices seem to be very different when it come to pairing negotiation. So different that they can be totally undetected by the coordinator depending of the FW version I use.

But something still eludes me. Most dongles are almost the same in terms of HW (same chips), nevertheless they behave quite differently depending of the FW version. Some are more or less foolproof, while other requires testing to find proper FW. So either chips integration plays a major role in this (trace impedance, PCB design...) or the TI chips used have variations (fake chips ?). It still is a mystery to me...

Now coming back to the Sonoff, I am pretty sure that it is an excellent device in terms of build quality, but in my opinion more work needs to be done to stabilize the current FW to make it usable. The fact that it works better with a previous version of FW is a good sign.

convicte, 2021-12-23:
I've tried both the 202103xx and 202112xx versions for the 2562P coordinator with no luck so far.

Anyone was able to make it allow new devices?
My whole network of 40+ devices moved easily and painlessly, but the 2 switches from IKEA and one from Sonoff did not, and I can't make them join for the life of me.

Thank you in advance!

iedex, 2021-12-23:
The latest develop version (20211217) works for me.

convicte, 2021-12-23:
In that case I have no idea what to do to troubleshoot it.

Enabling pairing results in no results being printed across   several devices. My logs are of course flooded with normal device published messages but nothing from the pairing unit.

I transitioned from c2531 where everything worked fine but the network started becoming unstable with 40 devices so I switched to 2652p where the issues persist. The network is much more stable and worked great until one device dropped out and I wasn't able to repair it, nor any other for that matter.

tb205gti, 2022-01-02:
Same issue here - updated from C2531 to the recommended CC2652 - unable to pair my Ikea Switches. Some of them have stopped working, and will not reconnect to the network. Tried to remove them and re-pair - but I cannot pair anything anymore. :/

convicte, 2022-01-02:
Please see https://github.com/Koenkk/zigbee2mqtt/issues/10339#issuecomment-1003693981

The issue is getting attention there and you may be experiencing the same thing as I am.

Cheers!

Tomk211, 2022-01-28:
I migrated as well from c2531 to 2652p (sonoff) and had issues that pairing doesn't work at the beginning (first day). Second try on  ext day allow me to repair. Another day later most of the devices stop reporting data. Just a few were rejoining... any idea? Already restarted/repowered. No log entry...

tb205gti, 2022-02-12:
Just to keep this alive - this is still an issue.

Koenkk, 2022-02-12:
Related to https://github.com/Koenkk/zigbee2mqtt/issues/10339; probably not somethign that can be solved in the coordinator fw; issues occurs because the cc2531 -> cc2652r isn't working 100%. 

As a workaround repair your routers starting with the ones closes to the coordinator.

RualdOrdelman, 2022-02-21:
I got some similair problems. Started this week with Z2M and bought a SonOff ZigBee 3.0 stick. First 5 Xiaomi devices where paired succesfully but can't pair any new after that. 

So I updated the SonOff firmware from fw 20210708 to fw 20211217, but after that I got the same problem. My Z2M give me some PAN erros, so I started with a fresh Z2m docker container. So I tried to repair the 5 devices that where connected the day before. Without succes. I can't pair of repair any devices. 

So I tried developer FW 20220219, but same problem. I can't see any messages in the debug log. So looks like all the pair messages are ignored or something. Any hint or something else what to do? I'm just started with Z2M, so start over and repair things isn't a problem.

RualdOrdelman, 2022-02-21:
> @RualdOrdelman could you provide me your data/database.db and data/coordinator_backup.json?

I've attached a link with all files interesting files. 
[https://drive.google.com/file/d/1kb4TXaW94ZSftFj0SNYYrqQ4YgntX7of/view?usp=sharing](url)

Currently only one Xiaomi temperature sensor is detected / paired. All other sensor can't be paired.

Koenkk, 2022-02-21:
@RualdOrdelman do you still have a cc2531 adapter? If yes, please sniff the traffic when pairing the device: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

RualdOrdelman, 2022-02-21:
@Koenkk Nope this is my first Zigbee device. Do you have a CC2531 adapter that you recommend and is not that expensive?

Koenkk, 2022-02-21:
@RualdOrdelman could there be another zigbee network running? Also, is the coordinator connected via a usb extension cable and not close to other devices? (ssds, wifi routers)

RualdOrdelman, 2022-02-21:
Nope. I've disconnect the existing Xiaomi Gateways that I was using before. Tomorrow my USB extension cable will be delivered, so I will test with that. 

Ps do you have some telegram / mail option so we can discuss some ideas without disturbing everyone in this channel :)
My telegram user is @Ruald and mailadres is ruald@ordelman.org.

RualdOrdelman, 2022-02-22:
@Koenkk shame on me. I've connected the USB Extension cable and all my problems are solved. Damm doesn't expect that the coordinator will be so sensitive. 

Thanks for your support and all the hard work for this project.

closed 2022-04-01 by github-actions[bot]

brandonb1987, 2022-04-15:
still a big problem for me. Unable to connect any devices