Koenkk/Z-Stack-firmware #339: CC2531 does not see IKEA lights

peetereczek, 2021-12-11:
Hello,

With latest firmware CC2531_SOURCE_ROUTING_20211116 and CC2531_DEFAULT_20211115 coordinator stopped seeing IKEA lights, there is no connection / no message being captured or sent.
What I have observed:
- devices started to report out of network avaliability state
- they can't be paired again
- Zigbee2mqtt reports timeout with error: "Data request failed with error: 'No network route' (205)" for already paired IKEA light bulbs

Issue is not applicable for IKEA remotes, nor other TuYa devices which I have in my network.

jonathanathe, 2021-12-16:
Hi, the same thing is happening to me, my Ikea bulbs stopped working when I updated to the latest version CC2531_SOURCE_ROUTING_20211116.

Landrash, 2021-12-30:
I'm running the same firmware but it works without a problem for my IKEA light's. I got 18 of them connected to my network currently.

Morphy99, 2022-01-03:
I have the same issue with CC2530 and source routing firmware. After upgrading from Z-Stack_Home_1.2 20201127/20201128 to Z-Stack_Home_1.2 20211115/20211116 on my main network everything stopped working as I have mostly IKEA bulbs as routers. After rolling back to a previous version the network was still failing and eventually after many hours I had to re-pair all my IKEA devices, even the remotes and dimmers. Everything else seemed to recover OK after rolling back firmware and OS snapshots. Routers including INNR bulbs, Sonoff ZB Mini and HUE bulbs. Aqara contact sensors needed a button press but they came back.  

Now I've had some time to set up another test network I've tried again and reproduced the same issue. IKEA bulb and Sonoff ZB Mini routers stop responding after updating the coordinator firmware. Aqara Contact sensor and button seem to work OK still though.

Here's my Z2M versions:

Zigbee2MQTT version
1.22.2 commit: 414c51f
Coordinator type
zStack12
Coordinator revision
20211116

Debug log [here](https://pastebin.com/4P3pUzh2)

Morphy99, 2022-01-03:
After writing this a few minutes later it's started working! How weird! Should still be an issue though as this shouldn't take so long to recover surely? In a larger network it may not recover with the stress of so many more devices...

[Here's](https://pastebin.com/SxzQJkZh) a continuation of the above debug log which shows both routers working again.

svante-jacobsen, 2022-01-24:
I made the same upgrade [a few days ago](https://github.com/home-assistant/core/issues/63273#issuecomment-1017442925) and the network took a long time to recover (I pretty much left it for dead and when I came back 2 hours later all devices - mix of IKEA's and 3-4 other brands - were all back).

I had A LOT of other issues at the same time so it's hard to know exactly what was what.

Morphy99, 2022-01-24:
@svante-jacobsen out if interest, how many end devices do you have? Considering trying this again on my production system some time when I have a few hours spare!

svante-jacobsen, 2022-01-24:
30 devices in total (of which 14 aren on battery, 17 are IKEA).

DigibitDK, 2022-02-06:
Same issue here. Can't pair Ikea-devices on CC2531 and CC2652P with v3-20211116 Coordinator firmware. They do not show up at all.
Older firmware do kind of work (Not always perfect, but it is possible)
Other devices than Ikea will pair just fine. Already paired Ikea do still work fine.

closed 2022-03-18 by github-actions[bot]