Koenkk/Z-Stack-firmware #341: Z-Stack router firmware for CC2652P/CC1352 that support control of RF transmit power output dBm level remotely

Hedda, 2021-12-20:
Feature request for a Z-Stack router firmware for CC2652P/CC1352 that can support controlling of RF transmit power output dBm level remotely just, maybe by presenting the transmission power output option on the Z-Stack router as via switches or dimmer?

If the router firmware supported that then could expose the attributes to applications like Zigbee2MQTT when the CC2652P/CC1352 based device is joined to Zigbee network and there allow the user to select the power output in dBm on those routers without reflashing.

ruimarinho, 2021-12-21:
For reference, some experimental firmware versions have been posted on https://github.com/Koenkk/Z-Stack-firmware/issues/273. The difference between the stable firmware and the one provided by @bkupidura (with +20dBm) is staggering: from LQI 7 to 80 in exactly the same spot.

Hedda, 2021-12-22:
> For reference, some experimental firmware versions have been posted on #273.

Should be noted that the router firmware in #273 is not configurable to differed TX power output levels, it is instead hardcoded into the firmware image to always use +20dBm TX power output amplification and can not be changed without flashing a other firmware image with a different configuration, which again is might be too strong/high for some scenarios and as such not always optimal for all individual users situations.

The initial request from the original post above remains as it would be great to have a Zigbee router firmware for CC2652P/CC1352P where the  TX power output could be reconfigured remotely.

WoLpH, 2022-01-02:
As a simpler solution for now, a normal / low power build and a high power build would already make a lot of difference.

digiblur, 2022-01-08:
I'd definitely drop some decent donations to @Koenkk  to make a CC2652 router firmware that had a Z2M converter with a slider of output power, restart button (probably needed to change power?), and LED state (on/off/status) picker.  Game changer indeed.

WoLpH, 2022-02-08:
@Koenkk would you accept a pull request with an extra high-power build in addition to the regular ones?

Koenkk, 2022-02-08:
> would you accept a pull request with an extra high-power build in addition to the regular ones?

No, this creates extra maintenance overhead for me. Instead this should be done by adding a manufacturer specific attribute on the genBasic cluster, from there it can be controlled from z2m by sending a write command.

wojciechczyz, 2022-02-25:
I can see there is already coordinator with 5.40 SDK in development branch, is there a plan for a router with 5.40? Thank you

Hedda, 2022-02-25:
> I can see there is already coordinator with 5.40 SDK in development branch, is there a plan for a router with 5.40? Thank you

Please start a new issue for that version request as that is off-topic for this specific subject or at least not directly related to topic.

WoLpH, 2022-03-28:
This if very much still relevant. Bumping

Hedda, 2022-04-28:
Still a relevant request I think if even technically possible?

ptvoinfo, 2022-04-28:
@Hedda 
Maybe, this is off-topic. I've implemented this feature in the premium version
https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/

Hedda, 2022-04-28:
> Maybe, this is off-topic. I've implemented this feature in the premium version
>
> https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/

Nice to know that it can technically be done at least. I guess chances are low that you PR a patch for it to this repo?

https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/firmware.patch

ptvoinfo, 2022-04-28:
> I guess chances are low that you code a patch and do a PR for it to this repo?

My changes are incompatible with this repo.

Hedda, 2022-04-28:
@ptvoinfo off-topic but if Z2M needs a converter then maybe you need to create a "quirk" for ZHA Device Handlers for that too(?) -> https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling -> https://github.com/zigpy/zha-device-handlers/blob/dev/README.md and might also need ZHA code as well to expose config in its UI(?) -> https://github.com/zigpy/zigpy/discussions/934

WoLpH, 2022-05-29:
This would still be very useful. Bump

WoLpH, 2022-06-29:
still very much needed

isabido, 2022-08-10:
HI friends, I am looking for a reliable router to expand the range of my SONOFF Zigbee 3.0 USB Dongle.

In view of the problems many sonoff devices such as the zbmini and s26 sockets are having, I have no idea which way to go.

Will it be the most reliable Philips HUE plugs as routers? https://www.zigbee2mqtt.io/devices/929002240401.html

Or maybe other SONOFF Zigbee 3.0 USB Dongle in router mode?

Thanks in advance!

Hedda, 2022-08-11:
@isabido your question is off-topic here, please post in Zigbee2MQTT discussions instead -> https://github.com/Koenkk/zigbee2mqtt/discussions

isabido, 2022-08-11:
Sorry Hedda! thanks for the recommendation.

Hedda, 2022-09-11:
Please do not stale.

cameronjust, 2022-09-23:
Is there much progress on this?

I had some minor issues with my Conbee II in the past and the general consensus in zigbee2mqtt forums was to use a "SONOFF ZigBee 3.0 USB Dongle Plus" with the CC2652P chipset.

So just today I got around to re-pairing all my devices and I was quite amazed that anything connecting to the Sonoff coordinator directly had an abysmal LQI <38 even when in the same room.

https://imgur.com/VbgcRXq

Whereas any device connecting via a router (Philips Hue bulbs and Ikea power points) had quality LQI > 200 even through walls.

After seeing this led me to this thread. How are people even using (and reccomending) these devices if they get such poor connectivity to devices in the same room?

I can see earlier in this thread an updated firmware with higher power settings baked in. However that was from March 2021. There has been a new firmware released in Feb 2022 so it would be kind of going backwards.

I'll stay with the Sonoff for now but if I get dropouts I'll probably give up and go back to the Conbee II which had very strong connections to devices directly connected.

Hedda, 2022-09-23:
@cameronjust Please understand that this specific feature request is only about "Zigbee Router" firmware for CC2652P/CC1352P.

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router

This specific feature request https://github.com/Koenkk/Z-Stack-firmware/issues/341 has as such not anything to with using CC2652P/CC1352P as a Zigbee Coordinator.

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

PS: Off-topic; Zigbee2MQTT can already configure "transmit_power" for CC2652P/CC1352P with Zigbee Coordinator firmware, see:

https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html

and discuss here instead:

https://github.com/Koenkk/zigbee2mqtt/discussions/8840

Hedda, 2022-10-24:
This feature is still wanted.

Smandurlo, 2022-11-01:
UP :-)

Koenkk, 2022-11-02:
This is now supported by the Z-Stack_3.x.0 router 20221102 https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin . This will be supported in the zigbee2mqtt dev branch later today.

closed 2022-11-02 by Koenkk

wolph, 2022-11-02:
Fantastisch @Koenkk, dankjewel!

Koenkk, 2022-11-02:
@wolph graag gedaan 😄

shaggee, 2022-11-16:
> This is now supported by the Z-Stack_3.x.0 router 20221102 https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin . This will be supported in the zigbee2mqtt dev branch later today.

Will it work also with the coordinator firmware?

Smandurlo, 2022-11-18:
but, how to use it for the router? We need to replace the firmware (is stable or dev) for the router and move to the dev branch? Where will we find the option to increase the dB?

Hedda, 2022-11-18:
> but, how to use it for the router? We need to replace the firmware (is stable or dev) for the router and move to the dev branch? Where will we find the option to increase the dB?

You first need to flash your CC2652/CC1352 based Zigbee Router device with Koenkk's Z-Stack 3.x.0 router firmware version 20221102 or later (currently in he `develop` branch) -> https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin

You will then need to use the very latest `dev` branch of zigbee2mqtt from today as as that `dev` branch  just got the converter for it less than 24-hours ago -> https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#linux

Lastly, you set the wanted dB as Zigbee attribute for transmit_power in BasicCluster for the device in order to increase or lower.

(The feature will not be available in Z2M master until it gets bumped zigbee-herdsman-converters v14.0.669 as a dependency).

For reference:

zigbee-herdsman converter was committed to the `master` branch of zigbee-herdsman-converters on the 2nd of November 2022 -> https://github.com/Koenkk/zigbee-herdsman-converters/commit/8fb1ba4c232e451af1e9953b8c476c2f4b2ef561 and that means that the needed converter should be available in zigbee-herdsman-converters v14.0.669 -> https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v14.0.669

zigbee-herdsman-converters v14.0.669 was bumped in Z2M yesterday https://github.com/Koenkk/zigbee2mqtt/commits/dev

TheJulianJES wrote instructions for how to enable it specifically in Home Assistant's ZHA integration (if supported in Zigbee Router device firmware) but it should be similar in Zigbee2MQTT -> https://github.com/zigpy/zha-device-handlers/pull/1886

Hedda, 2022-11-18:
> Maybe, this is off-topic. I've implemented this feature in the premium version
> https://ptvo.info/zigbee-configurable-firmware-features/rf-signal-level-control/

@ptvoinfo Can you extend the converter for [custom_devices_diy.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/custom_devices_diy.js) to expose `transmit_power` option  for your devices too? See:

https://github.com/Koenkk/zigbee-herdsman-converters/commit/8fb1ba4c232e451af1e9953b8c476c2f4b2ef561

Smandurlo, 2022-11-20:
Thanks @Hedda , then I will wait for the stable releases for both firmware and z2m

877dev, 2022-12-03:
This change made the z2m stable release this month. Are there instructions for setting the transmit power?
Can it be done dynamically via a MQTT message or does it need to go in configuration.yaml and restart z2m each time?
Thanks for any info!

Shamshala, 2022-12-03:
> This change made the z2m stable release this month. Are there instructions for setting the transmit power?
> Can it be done dynamically via a MQTT message or does it need to go in configuration.yaml and restart z2m each time?
> Thanks for any info!

Last time I checked it's already implemented in 1.28.4-1 (the latest stable release).

Shamshala, 2022-12-03:
@877dev Last time I checked it's already implemented in 1.28.4-1 (the latest stable release). Just check the device's **Exposes** section and there is the **transmit_power** slider.

877dev, 2022-12-03:
@Shamshala i don’t use HA, I wonder if there is a similar option in Z2M frontend interface. 
But it sounds like it can be altered dynamically which is great 👍

Shamshala, 2022-12-03:
@877dev **Exposes** is within the z2m frontend 😉

Smandurlo, 2022-12-03:
Is the firmware still in develop branch? I am not able to flash it with
Flash Programmer 2. I don't know what the problem is because I am able to
write the master release without a problem.

877dev, 2022-12-03:
Here is how I flashed mine recently, I wrote a little guide:

https://pastebin.com/i8akGRCy

Smandurlo, 2022-12-03:
with bsl-master I was able to flash the develop router firmware. WTH of a
difference!!! From 2-3 lqi at default (I made it on purpose), to 54 lqi
with +20 dBm !!!

877dev, 2022-12-04:
I am seeing the option to alter transmit power for the zzh routers within 'exposes':
[image]
[image]

But the zzh coordinator does not show in that list as of course it's not a device. So my original question remains, can coordinator transmit power be changed dynamically (either via front end or MQTT message), or does it need to go in configuration.yaml and restart z2m each time?

>

Hedda, 2022-12-04:
@877dev transmit power for Zigbee Coordinator is off-topic here, this specific issue is only about Zigbee Router. Please start a Zigbee2MQTT issue or discussion for your questions regarding Zigbee Coordinator configuration -> https://github.com/Koenkk/zigbee2mqtt/discussions

Smandurlo, 2023-01-02:
I have a silly problem. I have 2 sonoff zigbee stick as router. I tried the router dev firmware: 1 stick +20dB ok, the other one doesn't accept the setting.
I tried the 6.10.01.01 and again, the first stick is ok with +20dB, the other one no:

Publish 'set' 'transmit_power' to '0x00124b0024c0af49' failed: 'Error: Write 0x00124b0024c0af49/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 38967 - 242 - 100 - 0 - 4 after 10000ms)'

Any idea?

xekil, 2023-01-07:
> 

Same Problème, help please :)

s0ftice, 2023-02-03:
Same issue. I have 4 routers (all CC2652P - SONOFF Dongle-P) and around the Z2M 1.28.4 upgrade (early December), my 4th router suddenly started dropping connected devices. But not sure whether that was related to the Z2M upgrade. Anyways, I upgraded all router firmwares to 20221102 (by the way Z2M still shows 20220125 for two of them, even after re-pairing, seems to be a display bug).

Now, with all 4 routers on firmware on 20221102 and Z2M on 1.30.0 in the meanwhile, I am able to change the transmit_power (thanks for that!! @Koenkk), but my 4th router that was dropping connected devices also does not accept changing its transmit_power with the same error as @Smandurlo above. Simply times out after 10 seconds.
Same behaviour with the router close to the coordinator (LQI ~200).

I thought it's a hardware issue and ordered a replacement dongle. Upgraded to 20221102, but exactly the same issue.

Reading the report above and jumping to conclusions, seems to be an issue with people having more than one CC2652P...?

Any ideas? Thanks!

xekil, 2023-02-03:
@s0ftice i only have one such dongle and i have the same problem

Hedda, 2023-02-07:
@Smandurlo best is probably if you open a new ”bug” issue for tracking that issues as @Koenkk might miss it here since this ”feature request” issue as it is already closed.

Smandurlo, 2023-02-07:
@Hedda thanks for your suggestion. It is already open (https://github.com/Koenkk/zigbee2mqtt/issues/15912), but it didn't get any feedback from @Koenkk. Anyway, I will try to update my previous post. Thanks.