Koenkk/Z-Stack-firmware #343: Sonoff Zogbee 3.0 USB

stu1811, 2022-01-01:
I flashed the router firmware to the Sonoff Zigbee Dongle. I hoped it would would auto enter pairing mode when I plugged it into a USB outlet. ZHA did not find the device when I tried to add it. How do I enable pairing mode?

This is the output from flashing the latest dev firmware if it matters:
```
Opening port /dev/ttyUSB1, baud 500000
Reading data from ./CC1352P2_CC2652P_launchpad_router_20210128.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C0:AE:DF
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0xe0c256fd)
```

closed 2022-01-02 by stu1811

kris-anderson, 2022-02-09:
Did you end up getting this to work? I'm stuck on this part as well.

stu1811, 2022-02-09:
@kris-anderson Yes, enable pairing on ZHA/zigbee2mqtt and the connect dongle to power

kris-anderson, 2022-02-10:
@stu1811 

I'm doing something wrong then. Maybe it's how I'm flashing my stick. This is the command I'm running to flash the router firmware:

```shell
python Downloads/cc2538-bsl.py -evw -p /dev/tty.usbserial-210 --bootloader-sonoff-usb Downloads/CC1352P2_CC2652P_launchpad_router_20210128.hex
sonoff
```

Did you flash your device doing something differently?

stu1811, 2022-02-10:
`python3 cc2538-bsl.py -evw -p /dev/ttyUSB1 --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20211217.hex`

I'm on the `develop` branch

stu1811, 2022-02-10:
@kris-anderson I forgot, I'm using zigbee2mqtt edge

kris-anderson, 2022-02-10:
Thanks for the feedback @stu1811 . I'm looking for stability so I'm going to stay on the non-dev branch for now, but I'll keep checking each time my instance has an update available.

Thank you again!

Rxich, 2022-03-10:
> Did you end up getting this to work? I'm stuck on this part as well.

I too am stuck. Flashed the launchpad router firmware successfully, but it never will pair to my coordinator hub(Hubitat) I am able to pair many other zigbee routers but not this dongle.

ptvoinfo, 2022-03-11:
@Rxich launchpad may have different chips (CC2652P or CC2652R). Did you use a firmware for the corresponding chip?

NickUfer, 2024-09-02:
Did anyone ever figure this out?