Koenkk/Z-Stack-firmware #345: Flashed custom firmware with no BSL. Can I flash it without JTAG?

mortenmoulder, 2022-01-03:
Long story short: I'm goofing around with PVTO's custom firmware tool. I was too excited and I messed it up. I created a firmware with no BSL enabled (mistake number 1) and I also set P15 (FLASH button on the board) to "External pull-up" (mistake number 2). Without realising my mistake, I just flashed it and it worked perfectly.

So now I'm stuck with a CC2652P that is essentially bricked, because I don't have a JTAG device. I do have a CC Debugger if that's any help.

I have since then made a new firmware with SBL set to P15, and I also set a random unused pin to GPIO and pull-up, hoping I can trigger it by shorting it to GND.

Am I screwed or is there a way to recover? The dongle I have is a Z-Bee Duo.

mortenmoulder, 2022-01-03:
To answer my own question after 4+ hours of debugging: Yes. Using a Raspberry Pi 4, I was able (after some troubleshooting) to get OpenOCD to flash my CC2652P stick using JTAG. I followed a [Russian guide](https://www.youtube.com/watch?v=0XOozGlbU7U) (turn on subtitles), but it seems like there's another tutorial very similar [in English](https://gist.github.com/hwhw/fc43892785aa84913d03495c97b0f25a).

I kept getting connection errors, but at one point I magically got a connection. Turns out every 4th or 5th attempt worked, so I was just really unlucky to get errors all the time.

I used this command to flash Z-Stack back to the stick: 

```
sudo openocd -f interface/raspberrypi2-native.cfg -f target/ti_cc26x2.cfg -c "transport select jtag" -c "adapter speed 1000" -c "program zstack-firmware-file.hex verify reset exit"
```

After spamming that a few times, I finally got a great message:

```
** Programming Started **
Info : Padding image section 0 at 0x0000003c with 4 bytes
Info : Padding image section 1 at 0x00028bf4 with 4 bytes
Info : Flash write discontinued at 0x00029000, next section at 0x00057fa8
Warn : Adding extra erase range, 0x00056000 .. 0x00057fa7
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
```

and low and behold, it's recognized as a regular CC2652P stick with Z-Stack 3.x.0 on it again. Whew. What a night.

closed 2022-01-03 by mortenmoulder

XRyu, 2022-01-04:
Awesome! I can imagine that night! :) (had the same some years ago - flashing the wrong BIOS to a laptop, no graphics but booting just without the gpu chipset and building something to flash the correct bios without any screen from a USB - that was a ride) 

That said - It would be great if you could add a easy to follow article to the docs as this might render helpful for more people who get stuck on some point with more potential error sources, like right know with BSL, different kind of boards etc.

mortenmoulder, 2022-01-04:
> That said - It would be great if you could add a easy to follow article to the docs as this might render helpful for more people who get stuck on some point with more potential error sources, like right know with BSL, different kind of boards etc.

I think my comment covered most of the useful stuff. There are plenty of guides on getting OpenOCD working on a Raspberry Pi (the interface config file even tells you which pins to connect), and then it's just a matter of copying my one line of code and running it. No need to change any config files or anything like that, as the command does it all.

Not sure what else to explain. This was my first time playing with JTAG, and I got it reflashed in probably 2 hours, which also included booting my Raspberry Pi from a fresh SD card (where I spent 30 minutes trying to figure out why WiFi didn't work - turns out the wpa_supplicant file had the lines commented out, which I didn't notice at first).