Koenkk/Z-Stack-firmware #346: Sonoff Zigbee Dongle Plus Error

KrX3D, 2022-01-08:
Hello,

i got an Sonoff Zigbee Dongle Plus which i flashed with CC1352P2_CC2652P_launchpad_coordinator_20211217.hex and Flash Programmer 2 which worked.

But i cant get the Stick to work now, i tried it on

- Unraid Server -> Home Assistant VM
- Unraid Server -> IoBroker Docker
- Raspberry -> Home Assistant OS

but i always get the same error, here the one from Home Assistant:

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] config.sh: executing... 
[19:23:24] INFO: Both 'network_key' and 's0_legacy_key' are set and match. All ok.
[19:23:30] INFO: Virtual Machine not detected, enabling soft-reset
[cont-init.d] config.sh: exited 0.
[cont-init.d] structure.sh: executing... 
[cont-init.d] structure.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[19:23:34] INFO: Successfully send discovery information to Home Assistant.
2022-01-08T18:23:37.837Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2022-01-08T18:23:37.852Z DRIVER   version 8.9.2
2022-01-08T18:23:37.854Z DRIVER   
2022-01-08T18:23:38.059Z CONFIG   version 8.9.2
2022-01-08T18:23:41.115Z CNTRLR   querying controller IDs...
2022-01-08T18:23:42.951Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2022-01-08T18:23:44.074Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2022-01-08T18:23:46.233Z DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK 
                                  from the controller (ZW0200)
                                      at Driver.sendMessage (/usr/src/node_modules/zwave-js/src/lib/driver/Drive
                                  r.ts:3354:23)
                                      at ZWaveController.identify (/usr/src/node_modules/zwave-js/src/lib/contro
                                  ller/Controller.ts:677:33)
                                      at Driver.initializeControllerAndNodes (/usr/src/node_modules/zwave-js/src
                                  /lib/driver/Driver.ts:978:26)
                                      at Immediate.<anonymous> (/usr/src/node_modules/zwave-js/src/lib/driver/Dr
                                  iver.ts:858:16)
Error in driver ZWaveError: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:3354:23)
    at ZWaveController.identify (/usr/src/node_modules/zwave-js/src/lib/controller/Controller.ts:677:33)
    at Driver.initializeControllerAndNodes (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:978:26)
    at Immediate.<anonymous> (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:858:16) (ZW0100)
    at Immediate.<anonymous> (/usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:875:6)
    at processTicksAndRejections (internal/process/task_queues.js:95:5) {
  code: 100,
  context: undefined,
  transactionSource: undefined
}
Shutting down
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.`

dsantos21, 2022-01-08:
Are you sure those are the right logs (about zigbee)? Looks like they are pointing to a zwave problem.

KrX3D, 2022-01-09:
thanks for pointing that out. Home assistant showed zwave after i pluged the stick in

[image]

so i thought that this integration is needed.

now i installed Zigbee2MQTT and it works. Thanks

closed 2022-01-09 by Koenkk