Koenkk/Z-Stack-firmware #347: CC2652RB: does not work with CC2652RB_coordinator_20220103 or CC2652RB_coordinator_20211217

disaster123, 2022-01-15:
Hi there,

i've tried updating my slaesh CC2652RB stick from 20210708 to 20211217 or 20220103 but with 20220103 it does not even see any devices - with 20211217 it seems to see the devices but none reacts...

Stefan

Jost87, 2022-01-18:
Hi @disaster123,

i have updated my slaesh CC2652RB to 20211217 and it is working fine. If the devices (routers) do not respond, remove them from mains power for a few seconds as stated in the FAQ [https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices](url). That workes for me. 

Jost

disaster123, 2022-01-18:
mhm OK - might be an option but a bit strange... will try it in a few minutes... restoring old FW i got alle devices working again immediatly.

Might this be missing nvbackup or something like this?

I used:
```
c2538-bsl/cc2538-bsl.py -p /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_25_A0-if00-port0 -evw CC2652RB_coordinator_20211217.hex
```

for update.

disaster123, 2022-01-18:
Log says:
```
zigbee.1 | 2022-01-18 19:15:46.202 | warn | (813) DeviceAvailability:Failed to ping 0x00124b0022612054 GL-C-006
-- | -- | -- | --
zigbee.1 | 2022-01-18 19:15:46.118 | warn | (813) DeviceAvailability:Failed to ping 0x60a423fffeff1c35 TS011F
zigbee.1 | 2022-01-18 19:15:45.729 | warn | (813) DeviceAvailability:Failed to ping 0x84fd27fffe865141 TS0001
zigbee.1 | 2022-01-18 19:15:45.524 | warn | (813) DeviceAvailability:Failed to ping 0x84fd27fffe812d46 GL-C-006P
zigbee.1 | 2022-01-18 19:15:45.487 | warn | (813) DeviceAvailability:Failed to ping 0x84fd27fffe68e20b TS0001
zigbee.1 | 2022-01-18 19:15:45.369 | warn | (813) DeviceAvailability:Failed to ping 0x842e14fffe15d05e TS011F
zigbee.1 | 2022-01-18 19:15:45.228 | warn | (813) DeviceAvailability:Failed to ping 0x804b50fffe6b6c0b TRADFRI control outlet
zigbee.1 | 2022-01-18 19:15:45.115 | warn | (813) DeviceAvailability:Failed to ping 0xcc86ecfffe8db464 TRADFRI control outlet
```

disaster123, 2022-01-18:
OK it really fixes itself after 30 min.... i didn't expect to have to wait so long...

closed 2022-01-18 by disaster123