Koenkk/Z-Stack-firmware #348: CC2530 + CC2591 Can't join devices

kondas, 2022-01-20:
Hello!

I have a problem setting up a new zigbee network. I am using a WeBee Z0002 board, a CC2530+CC2591. I have a few Ikea bulbs, switches and remotes.

If i flash the "CC2530_CC2591_DEFAULT_20211115" coordinator firmware, I can't join any devices. Interview fails every time. I've attached logs from zigbee-herdsman of a failed attempt. Suprisingly when using "CC2530_DEFAULT_20211115" Firmware, devices can join succesfully, but the LQI is very poor, i guess because its not utilizing the CC2591.

Zigbee-herdsman debug logs:
`2022-01-20T16:32:07.040Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,217,121,0,34]
2022-01-20T16:32:07.041Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,217,121,0,34]
2022-01-20T16:32:07.041Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [217,121,0] - 34
2022-01-20T16:32:07.042Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":31193,"relaycount":0,"relaylist":[]}
2022-01-20T16:32:07.043Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:07.117Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,88,113,227,51,154,254,255,189,27,236,217,121,11]
2022-01-20T16:32:07.118Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,88,113,227,51,154,254,255,189,27,236,217,121,11]
2022-01-20T16:32:07.119Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [88,113,227,51,154,254,255,189,27,236,217,121] - 11
2022-01-20T16:32:07.121Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":29016,"extaddr":"0xec1bbdfffe9a33e3","parentaddr":31193}
2022-01-20T16:32:07.124Z zigbee-herdsman:controller:log Device '0xec1bbdfffe9a33e3' joined
2022-01-20T16:32:07.125Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:07.125Z zigbee-herdsman:controller:log Device '0xec1bbdfffe9a33e3' accepted by handler
2022-01-20T16:32:07.125Z zigbee-herdsman:controller:log Device '0xec1bbdfffe9a33e3' is already in database with different networkAddress, updating networkAddress
2022-01-20T16:32:07.137Z zigbee-herdsman:controller:log Interview '0xec1bbdfffe9a33e3' start
Zigbee2MQTT:info  2022-01-20 16:32:07: Starting interview of '0xec1bbdfffe9a33e3'
Zigbee2MQTT:info  2022-01-20 16:32:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe9a33e3","ieee_address":"0xec1bbdfffe9a33e3","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-01-20 16:32:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe9a33e3"},"type":"pairing"}'
2022-01-20T16:32:07.167Z zigbee-herdsman:controller:device:log Interview - start device '0xec1bbdfffe9a33e3'
2022-01-20T16:32:07.175Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29016,"nwkaddrofinterest":29016}
2022-01-20T16:32:07.175Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,88,113,88,113,35]
2022-01-20T16:32:07.194Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2022-01-20T16:32:07.194Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2022-01-20T16:32:07.195Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2022-01-20T16:32:07.195Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2022-01-20T16:32:07.195Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:17.203Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '29016' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2022-01-20T16:32:17.204Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 29016
2022-01-20T16:32:17.206Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":29016,"options":0,"radius":30}
2022-01-20T16:32:17.208Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,88,113,0,30,83]
2022-01-20T16:32:17.224Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2022-01-20T16:32:17.225Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2022-01-20T16:32:17.226Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2022-01-20T16:32:17.226Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2022-01-20T16:32:17.226Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:20.229Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":29016,"nwkaddrofinterest":29016}
2022-01-20T16:32:20.229Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,88,113,88,113,35]
2022-01-20T16:32:20.248Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2022-01-20T16:32:20.249Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2022-01-20T16:32:20.249Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2022-01-20T16:32:20.249Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2022-01-20T16:32:20.250Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:29.614Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.616Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.617Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [217,121,0] - 34
2022-01-20T16:32:29.619Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":31193,"relaycount":0,"relaylist":[]}
2022-01-20T16:32:29.621Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:29.624Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.624Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.627Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [217,121,0] - 34
2022-01-20T16:32:29.627Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":31193,"relaycount":0,"relaylist":[]}
2022-01-20T16:32:29.628Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:29.721Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,88,113,227,51,154,254,255,189,27,236,217,121,11]
2022-01-20T16:32:29.721Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,88,113,227,51,154,254,255,189,27,236,217,121,11]
2022-01-20T16:32:29.722Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [88,113,227,51,154,254,255,189,27,236,217,121] - 11
2022-01-20T16:32:29.722Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":29016,"extaddr":"0xec1bbdfffe9a33e3","parentaddr":31193}
2022-01-20T16:32:29.723Z zigbee-herdsman:controller:log Device '0xec1bbdfffe9a33e3' joined
2022-01-20T16:32:29.723Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:32:29.724Z zigbee-herdsman:controller:log Device '0xec1bbdfffe9a33e3' accepted by handler
2022-01-20T16:32:29.725Z zigbee-herdsman:controller:log Not interviewing '0xec1bbdfffe9a33e3', completed 'false', in progress 'true'
2022-01-20T16:32:29.731Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.732Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,217,121,0,34]
2022-01-20T16:32:29.732Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [217,121,0] - 34
2022-01-20T16:32:29.733Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":31193,"relaycount":0,"relaylist":[]}
[... 186 lines cut ...]
2022-01-20T16:34:12.625Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2022-01-20T16:34:12.625Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2022-01-20T16:34:12.627Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-01-20T16:34:15.631Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38051,"nwkaddrofinterest":38051}
2022-01-20T16:34:15.632Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,163,148,163,148,35]
2022-01-20T16:34:15.648Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
[... cut ...]
Zigbee2MQTT:error 2022-01-20 16:34:26: Failed to interview '0xec1bbdfffe9a33e3', device has not successfully been paired`

Also, i have a CC2531 USB stick. I was able to set up everything about a year ago using 1.2HA coordinator FW rev "20201127..." All devices were working properly, also some group binding was utilized. But now the issue is similar with that stick also, no devices can join /finish interview using CC2531_DEFAULT_20211115 (1.2HA) firmware. 

I tried multiple host machines (linux, arm), changing USB ports, re-flashing the firmware multiple times to no avail.

schlaubstar, 2022-01-22:
Hi  kondas, I did not look into the logs at all but I encountered a similar issue tonight.
At least in my case I could identify the issue: I powered the device (CC2530+CC2591) the identical way I tried with the CC2530 alone, using a "mobile charger" with an attached "CP2102 CH340G USB TO TTL PL2303 CP2102 USB to UART" Adapter and taking the power from the 3.3V port. Obviously the combination of CC2530 + CC2591 requests way more power than the USB-Adapter seems to reliably deliver. When using a serious power supply all issues were gone and the new router behaved as reliable, as the old one. Maybe, you can also give it a try.

kondas, 2022-01-22:
Hi! Thanks for the help. I've tried powering the board from 3.3v 3A regulator but the problem persists. I'm trying to use the board as a coordinator, but every interview fails when using the correct firmware for the board.

closed 2022-03-01 by github-actions[bot]