Koenkk/Z-Stack-firmware #349: slae.sh/CC2652RB router firmware not pairing

errrrata, 2022-01-21:
I have 2 x `slae.sh CC2652RB`, and tested them successfully with the coordinator image, and decided to use one of them as router. Followed same instructions as for the coordinator firmware flashing, replacing the firmware with https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC2652RB_router_20210128.zip, the firmware gets upload without any issues, switched `zigbee2mqtt` in pairing mode (with 2 meters distance between them), however there is no pairing attempt in the `zigbee2mqtt` logs. I have attempted to factory reset the `CC2652RB` as well by pressing the button further from the USB, however there's no sign of reset or pairing, the green led stays on all the time, and no other change is visible.

errrrata, 2022-01-23:
Since creating this ticket, I decided to try replacing the dongles, and using the other dongle with the router firmware seems to work correctly. So basically, while both dongles work as coordinator, only one of them seems to work with the router firmware. Can it be a hardware issue considering both of them work as coordinator?

wizake, 2022-01-27:
In my case its a bit different.
I've used slae.sh stick as a coordinator, but repurposed it as a router when I got another model (ZigStar). 
The issues I had with it were: devices didnt seem to want to connect to it, and even if they did, after a while it dropped from the network along with the devices.. So at the moment it gathers dust on a shelf.

Koenkk, 2022-01-28:
I've released a new fw: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/router/Z-Stack_3.x.0/bin , could you check if the issue still occurs with this fw?

FreunderNacht, 2022-01-28:
Hello,
first of all, thank you for the work on the firmware.

I have the same problem.
I have a Sonoff Zigbee 3.0 USB dongle as a coordinator.
This works perfectly.

Now I got myself another 3 Sonoff.
I wanted to use these as routers, so I installed the router firmware.

However, the pairing attempts fail, I don't even see an attempt to pair in the Zigbee adapter of the IO Broker.

More people have this problem in the IO Broker Forum:

[Link IO Broker Forum](https://forum.iobroker.net/topic/48063/sonoff-zigbee-3-0-usb-dongle-plus/327?_=1643403351281)

I will try the new firmware tomorrow.

FreunderNacht, 2022-01-29:
I have now installed the develop firmware on the stick and tested it.
Unfortunately, it still doesn't work.

You can see the time running through the Zigbee adapter, but no attempt at pairing is visible. Nothing happens.

**But here is my procedure again:**
1. Stick flashed with new firmware,
2. disconnected from the power supply (as I have no Zigbee signal on the PC),
3. plugged it into a USB power supply near the coordinator,
4. waited 2 seconds,
5. disconnected again,
6. waited 2 seconds, plugged the stick back in,
7. started pairing mode.

**My setup:**
> Intel NUC with Proxmox
> IO Broker as VM
> Zigbee adapter (version 1.6.6)
> Sonoff Zigbee 3.0 Dongel Plus as coordinator (firmware 20210708)

I have already taught a few devices to the coordinator (Aquara motion detector, Aquara button, Sonoff door/window contact), which worked without problems. Only the routers (Sonoff Zigbee 3.0 Dongel Plus with firmware 20220125 or 20210128) do not work.

FreunderNacht, 2022-01-29:
I have now updated the coordinator to the latest firmware (20211217).
Now the pairing has worked as it seems:

<html>
<body>

zigbee.0 | 2022-01-29 14:47:43.682 | warn | Stopping to ping 0x00124b0024c94e62 ti.router after 10 ping attempts
-- | -- | -- | --
zigbee.0 | 2022-01-29 14:46:56.768 | info | Zigbee: stop joining
zigbee.0 | 2022-01-29 14:46:33.741 | info | DeviceConfigure successful 0x00124b0024c94e62 ti.router
zigbee.0 | 2022-01-29 14:46:33.660 | info | Configuring 0x00124b0024c94e62 ti.router
zigbee.0 | 2022-01-29 14:46:33.659 | info | Device '0x00124b0024c94e62' is supported, identified as: Custom devices (DiY) Texas Instruments router (ti.router)
zigbee.0 | 2022-01-29 14:46:33.658 | info | Successfully interviewed '0x00124b0024c94e62', device has succesfully been paired
zigbee.0 | 2022-01-29 14:46:10.220 | warn | Device '0x00124b0024c94e62' announced itself
zigbee.0 | 2022-01-29 14:46:10.169 | info | Starting interview of '0x00124b0024c94e62'
zigbee.0 | 2022-01-29 14:45:55.499 | info | Zigbee: allowing new devices to join.

</body>
</html>

**Here again is my current set-up:**
> Intel NUC with Proxmox
> IO Broker as VM
> Zigbee adapter (version 1.6.6)
> Sonoff Zigbee 3.0 Dongel Plus as coordinator (firmware 20211217)
> Sonoff Zigbee 3.0 Dongel Plus as router (firmware 20220125)

[image]

[image]

joselcaguilar, 2022-02-07:
I'm using the CC1352P2_CC2652P launchpad as a coordinator (fw 20220103) and slaesh.sh/CC2652RB router (fw 20220125) and everything work fine with ZigBee2Mqtt 1.23.0, however, when I click on Router CC2652RB device in Z2M, it still show the old fw version installed (20210128) instead of the new ones (20220125), and I'm pretty sure that the new fw was flashed successfully.

Have you experienced this?

Koenkk, 2022-02-07:
@joselcaguilar z2m doesn't read the new firmware version automatically. You can read it manually via the frontend:

[image]

lsilva171, 2022-02-28:
I am using Electrolama as my coordinator with 20211217 firmware and slae.sh/CC2652RB with 20220125 firmware. I can say it paired successfully all times I tried and its correctly identified as a router. I also checked the firmware version via frontend and it shows 20220125.

However, it stops responding every time (or never responds) and eventually shows offline. Another interesting quirk is it drops out of the network if I press the BL button (further from USB connector) after showing offline for some time.

johare279, 2022-03-12:
I have sonoff dongle plus coordinator with 20211217 firmware and another sonoff dongle plus with 20220125 router firmware and still cannot discover the router. Am I missing something?

johare279, 2022-03-12:
> I have sonoff dongle plus coordinator with 20211217 firmware and another sonoff dongle plus with 20220125 router firmware and still cannot discover the router. Am I missing something?

update (to help other newbies like me): I bought a few sonoff usb plus to use as routers. Turns out the first one I tried was actually faulty. But, I did learn that you need to keep the flased router in the host machine and use the z-tool (https://www.ti.com.cn/tool/cn/SIMPLELINK-CC13X2-26X2-SDK) to set the baud rate to 115200 (go to settings, select the COM port that stick is on, edit, and set rate - ok - apply - ok). you then need to power down the stick and start the pairing on zigbee2mqtt and then power up the stick (still in host machine - computer you flashed from). after the pairing is totally complete, you can power down the stick and transfer the stick to whatever usb power supply you want. I may not be doing it the way it is supposed to be done - but this worked for me.

closed 2022-04-19 by github-actions[bot]

wizake, 2022-12-30:
Decided to give it another try after almost an year. Latest router fw. Same issue with the slae.sh stick as a router: goes offline after a while