Koenkk/Z-Stack-firmware #353: Support for CC2652R7/P7

io53, 2022-02-11:
You can now buy CC2652R7 devkit, the chip has more flash and ram than previous versions: https://www.ti.com/tool/LP-CC2652R7

If that is something zigbee benefits from it would be great to see it supported :).

digiblur, 2022-02-11:
Looks like xmit power is back to 5 on this one?

digiblur, 2022-02-11:
ahh...that is good to know.  I was looking at the sheet you linked.

io53, 2022-02-15:
I received my CC2652R7 devkit today and was able to port of the changes from the other chips and flash it. Works fine with z2m. 
I don't know how to take advantage of the bigger memory yet tho, got some reading to do :)

closed 2022-03-25 by github-actions[bot]

Sheng2216, 2023-04-11:
> I received my CC2652R7 devkit today and was able to port the changes from the other chips and flash it. Works fine with z2m. I don't know how to take advantage of the bigger memory yet tho, got some reading to do :)

Hi, I am trying to build firmware for the CC2652R7 and test it with Z2M, we plan to use it for Matter later. But we got stuck at porting the firmware.patch to R7, do you have any suggestions on where to get started?

io53, 2023-04-13:
Was over a year since did this but if I remember correctly I just followed the instructions but also added the LP_CC2652R7. After applying the patch I looked at the diff for R1 and manually copied over the changes to R7.

Are you planning to use zigbee-herdsman and basically replace the mqtt in z2m with matter?

Sheng2216, 2023-04-13:
> Was over a year since did this but if I remember correctly I just followed the instructions but also added the LP_CC2652R7. After applying the patch I looked at the diff for R1 and manually copied [...]

We are not there yet. Right now we are only working on making it work with Z2M first. We copied the changes to R7 but we got some error message about "network commissioning failed", we are still working on it.

> Zigbee2MQTT:info  2023-04-13 12:02:44: Logging to console and directory: '/app/data/log/2023-04-13.12-02-44' filename: log.txt
> Zigbee2MQTT:info  2023-04-13 12:02:44: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
> Zigbee2MQTT:info  2023-04-13 12:02:44: Starting zigbee-herdsman (0.14.96)
> Zigbee2MQTT:error 2023-04-13 12:03:27: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2023-04-13 12:03:27: Failed to start zigbee
> Zigbee2MQTT:error 2023-04-13 12:03:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> Zigbee2MQTT:error 2023-04-13 12:03:27: Exiting...
> Zigbee2MQTT:error 2023-04-13 12:03:28: Error: network commissioning failed - timed out waiting for nib to settle
>     at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:364:19)
>     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
>     at Zigbee.start (/app/lib/zigbee.ts:58:27)
>     at Controller.start (/app/lib/controller.ts:101:27)
>     at start (/app/index.js:107:5)

Sheng2216, 2023-04-21:
Thank you for your suggestion, we manually port the changes without changing anything and the firmware works smoothly without any issues. We created a repo for future use: https://github.com/RAKWireless/CC2652R7_firmware

Sheng2216, 2023-04-21:
> Was over a year since did this but if I remember correctly I just followed the instructions but also added the LP_CC2652R7. After applying the patch I looked at the diff for R1 and manually copied [...]

Thank you for your suggestion, we manually port the changes without changing anything and the firmware works smoothly without any issues. We created a repo for future use: https://github.com/RAKWireless/CC2652R7_firmware