Koenkk/Z-Stack-firmware #357: Restore original firmware back to sonoff zb dongle?

terslowe, 2022-02-19:
I have installed the itead firmware with your tutorial but i need to restore the original firmware of sonoff. I didn't maked a backup. How can i restore back to it?

atulalvenkar, 2022-02-19:
If you are talking about sonoff zigbee dongle, they come with this same zstack firmware. They don't have their own firmware for zigbee coordinators. Just that it might be some older version.

terslowe, 2022-02-19:
My problem was with the shipped firmware Paulman leds was working/pairing with ZHA. But after flashed the firmware i can't find paulman by pairing.

juppyjupp, 2022-03-10:
I guess you're talking about "SONOFF Zigbee 3.0 USB Dongle Plus"
This donge ist usually shipped with Coordinator Version 20210708
FW CC1352P2_CC2652P_launchpad_20210708.zip

drthanwho, 2022-03-15:
I can't verify that as well since I recently bought one and it has the 20210708 version on it

Hedda, 2022-04-07:
The very first batch came pre-flashing with older firmware from 20210120 but can still get all old firmware via commit history:

https://github.com/Koenkk/Z-Stack-firmware/commits/master/coordinator/Z-Stack_3.x.0/bin

More info on "SONOFF Zigbee 3.0 USB Dongle Plus" here:

https://community.home-assistant.io/t/iteads-sonoff-zigbee-3-0-usb-dongle-plus-based-on-texas-instruments-cc2652p-20dbm-radio-mcu-now-sold-for-14-99/340705

closed 2022-05-15 by github-actions[bot]