Koenkk/Z-Stack-firmware #361: Z-Stack_Home_3.x.0_20211217 not working with EBYTE E79-900DM2005S CC1352P module

LDenissen, 2022-03-06:
I have made my own dongle using an EBYTE E79-900DM2005S CC1352P module and this has been working flawlessy so far, only with the new 3.x.0 20211217 firmware I only get "Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel". After flashing the dongle back to the previous 3.x.0 20210708 firmware everything works like usual.

I suspect it could be the Sub-1GHz RF switch on the module that is controlled by DIO_5 and DIO_6, that some behaviour changed here for setting the LEDs.

egony, 2022-03-09:
You can not use power amplifier on this module, so use transmit_power 5 or less.

LDenissen, 2022-03-09:
I doubt this is the issue, as with the older firmware I can set  transmit_power to 9 without any problems

LDenissen, 2022-03-12:
Have to say you are right, but this also means something in the firmware is not handled very well.
Downloaded CCS and applied the patch to the example project, then changed the default TX power from 9 to 5 and now it works.

So setting the TX power to 9 in the older firmware should also not work, but does not seems to give any issues/errors.
With the more recent firmwares that use TX power 9 as default, then even if you put it back to 5 in the config the module won't start at all. Maybe better to just stick with the default of TX power at 5 in the firmware, as it is breaking compatibility with the CC1353P modules. For other modules it is still possible to increase the TX power in the config file.

Koenkk, 2022-03-13:
The old firmware didn't support 9 TX power, therefore it probably defaulted to 5. I expect you get the same result when setting 20 for the old firmware.

closed 2022-04-20 by github-actions[bot]