Koenkk/Z-Stack-firmware #362: Coordinator Firmware speed comparisons

nickwild-999, 2022-03-10:
I am trying to compare different coordinator firmware to optimize my sub-optimal LQI's  - I am using a test rig with RPI running Z2M only in a 15ft x 15ft room.

I have ongoing problems with Sonoff sockets losing connections so I have omitted those and just focussed on 4 Ikea Bulbs 2 of which are only 6 to 10 ft away from the dongle which is on an extension cable

To be honest there seems little difference in performance between firmware's set to different output powers and amplified or unamplified 2652s.

Does anybody have any thoughts on how these can be improved 

[image]

Koenkk, 2022-03-10:
> I have ongoing problems with Sonoff sockets losing connections 

Might be interesting for you: https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/

nickwild-999, 2022-03-10:
Thanks @Koenkk yes I already had a dialogue with @ptvinfo about that. I am going to try flashing this over the weekend and hope it solves things

nickwild-999, 2022-03-10:
@Koenkk actually once thing I have noticed after I posted this is that there is a big disparity between the LQIs on the map and on the dashboard. Looking at the Sonoff dongle with fw `20220219` the map is showing 250 - 255 on the maps but only 120 on the dashboard.

Do you have any thoughts on the reason for that? And should I stop optimizing and start living! :-)

Koenkk, 2022-03-10:
@nickwild-999 

- Dashboard: LQI from last hop is shown, e.g. Sonoff -> router A -> coordinator, the LQI of router A -> coordinator is shown. The zigbee message received by the coordinator only contains this LQI information.
- Network map shows the LQI of a [LQI request](https://www.digi.com/resources/documentation/Digidocs/90002002/Concepts/c_zb_example1_send_ZDO_LQI_request.htm?TocPath=Transmission%2C%20addressing%2C%20and%20routing%7CZDO%20transmissions%7CReceiving%20ZDO%20command%20and%20responses%7C_____1)

nickwild-999, 2022-03-10:
@Koenkk Thanks for explaining. So in fact in my case the second number in the map corresponds to the Dashboard LQI.

Do you have any thoughts  about the lack of difference between the LQIs between the firmware/powers - is this because this actual dashboard number is the LQI FROM the device to the coordinator?

Thanks again for all your work on Z2M

Koenkk, 2022-03-10:
@nickwild-999 because transmit power doesn't improve LQI from device -> coordinator, you can shout harder but this won't make your ears better

nickwild-999, 2022-03-10:
Thanks @Koenkk yes that is what I now realise with your explanation. 📣 Thanks again

charettepa, 2022-03-11:
> Thanks @Koenkk yes I already had a dialogue with @ptvoinfo about that. I am going to try flashing this over the weekend and hope it solves things

Please advise on results, and flash method ie. (OTA or USB Serial)
I have the same issue with zbmini

charettepa, 2022-05-02:
@nickwild-999 
did you flash the firmware from https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/
did it help?
what flash method did you use?

looking for help as i have that device with the same issue

@Koenkk 
this particular device is giving me issues since 20220219
whereas on 20211217 it was stable even though the signal was weaker

nickwild-999, 2022-05-06:
@charettepa 
> did you flash the firmware from https://ptvo.info/custom-firmware-for-sonoff-zbmini-01minizb-and-ewelink-switch-zr-02-923/
> did it help?

I didn't have time and ended up buying ikea sockets that just worked. I do have loads of ZBMINIs that I do need to flash, but the 3d printed jig I have does not fit and I don't have time to amend ATM

mrmaximas, 2024-01-10:
> because transmit power doesn't improve LQI from device -> coordinator, you can shout harder but this won't make your ears better

@Koenkk what does lqi mean in case of a link between the router and coordinator? i have some routers probably can't hear coordinator well. can the coordinator tell the router to use not a direct route but through other routers when it would be better?

Koenkk, 2024-01-11:
No that's not possible.

mrmaximas, 2024-01-11:
> No that's not possible.

how does the network decide which route packets take?

Koenkk, 2024-01-11:
The routers decide