Koenkk/Z-Stack-firmware #363: smlight-slzb-05 "Update Failed: Wrong Magic Byte"

blackie333, 2022-04-13:
Hello,
I'm trying to update Zigbee firmware in my "SMARTLIGHT Zigbee LAN Adapter CC2652P Model SLZB-05" to the latest version "CC1352P2_CC2652P_other_coordinator_20220219.hex" via adaptor web interface but every attempt ends with an error: "Update Failed: Wrong Magic Byte". 
Restart of the adaptor or Toggle to flashmode before flashing has no effect, still the same result.
Bought this model because of easy flashing via web, but my first attempt leads to problems.
Not sure what else to try, please advice me the correct procedure. Thnx

giovbolo, 2022-04-14:
I have the same problem  when flashing  the Zigbee firmware version "CC1352P2_CC2652P_other_router_20220125.hex" in my "SMARTLIGHT Zigbee LAN Adapter CC2652P Model SLZB-05" . --> "Update Failed: Wrong Magic Byte"

blackie333, 2022-04-17:
Thank you for the tips. I tested "ZigStar GW Multi tool", but it was unable to find/connect with the module via IP address. Haven't tested the other tools for USB connection yet, not sure the USB port on the module is connected to zigbee chip.
According to module description USB connector is there just to supply power to the adapter (no hybrid mode mentioned). I'm also a bit afraid to use bootloader flashing tools (to not kill module web/zb functionality).

kireque, 2022-05-10:
Follow this guide:
https://smartlight.me/diy-smart-home/slzb05-flashing

Don't forget to stop the zigbee2mqtt container, or you will get an error during connecting to the adapter.

blackie333, 2022-05-29:
Thanks, I have tried to follow the guide but there is no "START" button on the screen, only READ, WRITE, ERASE, READ_IEEE buttons and similarly named checkboxes. Tried the READ button to backup old firmware, but "unknown error" occurred, see the screenshot
[image]
.

smaicloud, 2022-06-17:
Remove the port, read ieee must show the ieee in the box, check erase, write, verify and the start button appears. FW flash is successful.

blackie333, 2022-06-20:
Thank you very much man. Had some problems with "Read IEEE" but after stopping Z2MQTT+restarting LAN the connection started to work. Flashed the f/w [20220219] successfully.

After re-starting Z2MQTT I can see that LQI value of all my routers seems way below it was with previous (12/2021) firmware.
Reported LQI value fluctuates between 29-36 (3x Philips Hue bulbs, 3x Lidl plugs, 2x OpenLumi routers), even for those which are just 5m away from the coordinator.

closed 2022-07-29 by github-actions[bot]