Koenkk/Z-Stack-firmware #365: Problem pairing a device from Owon/CasaIA

pipiche38, 2022-04-28:
I'm struggling to get a PIR323 or a CMS323 from Owon.

It looks like the verify key is not coming on time and so the device leave the network.
Here is a sniff .

Transaction key: 82315266be0b67626ac22107f2da3150

[CMS323-Pairing-Failing-on-Sonoff.pcapng.zip](https://github.com/Koenkk/Z-Stack-firmware/files/8581047/CMS323-Pairing-Failing-on-Sonoff.pcapng.zip)

pipiche38, 2022-05-03:
@Koenkk did you already experienced this type of issue with battery based devices ?

Koenkk, 2022-05-03:
I think this happens because other commands are queued before it. Everytime the device does a `Data Request` one of the queued commands is being send, if too many commands are queued before it the `Confirm Key` will come too late. I think the zigpy-znp adapter has to be changed here to wait a bit before sending all kind of commands to the device when it joins.

pipiche38, 2022-05-03:
@puddly what do you think ? I have a similar issue (less often) with bellows as well .

puddly, 2022-05-03:
I find it strange that the device announces itself on the network *before* verifying its key. It even completes the basic device interview and binding before verifying the key.  I can't find anything in the spec referencing the order in which this is supposed to happen but this seems wrong.

zigpy-znp has logic to wait a few seconds for devices to announce themselves after being notified of the key exchange by Z-Stack (since some devices do not communicate at all until they send the announce), but it is unexpected for a device to interleave the two processes.

I'm not sure how much of zigpy's initialization code you're using (if any) but can you try Koen's suggestion of [adding a delay before initializing this device](https://github.com/zigpy/zigpy/blob/64c2a288b583e0a756f365bd29aa57f70d64415d/zigpy/device.py#L175)? Maybe you can match it by an IEEE address prefix.

pipiche38, 2022-05-03:
I'll try that approach, and will let you know.

pipiche38, 2022-05-03:
Ok. 6 seconds wait after sending the 1st command.
It looks like they start the Key Validation after the first command.

The strange thing is despite the behaviour looks strange, I didn't see this issue on Zigate .

closed 2022-05-03 by pipiche38

pipiche38, 2022-05-05:
@puddly I got a response from Owon why they are not doing it during the Association time. In fact they wait for the Node Descriptor response from the Coordinator to know which Zigbee stack they are running. if the Zigbee stack is 21 or 22 and above they will do the Key Verification to comply with ZB 3.0

The shame is that we do not see the Node Descriptor Request, nor cannot change the Node Descriptor response (fully embedded into the firmware)

pipiche38, 2022-05-05:
Yes the device does the Node Descriptor request, and I think that from what I saw up to now, most of the devices does that. 

Like the Aqara, which then stop working if the Node Descriptor Answer do not provide a Aqara/Xiaomi manufacturer code ;-)

puddly, 2022-05-05:
Let's move this conversation back to https://github.com/zigpy/zigpy-znp/issues/149, it doesn't seem related to the Z-Stack firmware itself if it can be solved by adding delays within the radio library.