Koenkk/Z-Stack-firmware #367: Incorrect CC2652R firmware flashed on SONOFF Zigbee 3.0 USB dongle

leloch, 2022-05-02:
Somehow I managed to flash `CC2652R_coordinator_20220219.zip` firmware on  SONOFF Zigbee 3.0 USB, rather than `CC1352P2_CC2652P_launchpad_coordinator_20220219.zip` one...

Now, the device is rather bricked, the serial port is visible - but there is no response when using any of the flash tools (this one as well as Flash Programmer 2) and the 'boot' button does not seem to do anything either...

What are my recovery options? I have got FT232 JTAG cable, would that one be good enough to re-flash recover it?
Possilby some other method?

Or am I pretty much screwed and should just but another unit?

Koenkk, 2022-05-02:
Should be doable via a JTAG flasher and https://www.ti.com/tool/FLASH-PROGRAMMER

leloch, 2022-05-03:
I only got a regular FT232 JTAG cable, it looks like these TI chips require a 'special' cJTAG interface (cc-debugger?).
I could not locate any resource explaining how to connect regular JTAG to this thing - any hints would be very much apperciated.

petergebruers, 2022-05-03:
Yes it is cJTAG, not compatible with JTAG. I haven't needed this myself yet... but I know this competing product is based on a nearly identical chip an the maker describes how to flash it:

https://electrolama.com/radio-docs/advanced/flash-jtag/

Basically, either you need a JLINK programmer, or one of the TI development boards which have a XDS110 onboard. I own both. He also mentions a third option, the CC-DEVPACK-DEBUG, I do not have that one.

leloch, 2022-05-03:
Thank you - this is what was afraid of. 
I don't have either and it does not really make sense to get the adatper just to fix this one specific unit, so in such case I will just get another dongle, luckily these are quite cheap (still). The borked one will go into the drawer, maybe one day I will have the kit needed to fix it :)

closed 2022-05-03 by Koenkk

kaiekaie, 2022-11-04:
I did this with the ZB dongle-E and used a wire to jump the boot and ground pins to enter bootmode. Maybe you can try  touching any of connectors on the (P board) close to the boot button? to see if it starts in Bootmode.

tobiasfrejo, 2022-11-04:
I have been able to use a Raspberry Pi with OpenOCD as a JTAG debugger to flash the correct firmware to a generic CC2652P dongle. Though that one does have through-hole soldering points that exposes the remaining pins for a standard JTAG connection, witch it doesn't seem like the SONOFF dongle has. 
[image]

I have done a write-up about it in case anyone stumbles upon the same problem with these tools available. 
https://gist.github.com/tobiasfrejo/d98e9f4cbbc6893662ab032b06388685

michaelp1742, 2022-12-14:
Tobias, thank you for researching and documenting this. 
I bricked my Sonoff ZBDongle-P (CC2652P) by programming it immediately after my Electrolama zzh! (CC2652R). I managed to tack solder directly to the two IC pads (0.5mm spacing) with fine multi-stranded wire from an old USB cable using an iPhone as a magnifier.
The only change required was to drop the "adapter speed" to 500 to get a reliable JTAG connection.

BoBeRzE, 2023-12-12:
@michaelp1742 I have also bricked my Sonoff ZBDongle-P. Can you please give me some details which soldering points you used? I would also try to get the device up and running again with the help of the Raspi.