Koenkk/Z-Stack-firmware #372: Firmware for Sonoff Zigbee Bridge Pro

s-hadinger, 2022-05-21:
The new Sonoff Zigbee Bridge Pro has ESP32 + CC2652P. I had issue with the onboarded ZNP 2.7.11, so I flashed back your firmware `CC1352P2_CC2652P_launchpad_coordinator_20220219.hex`.

The main difference with SONOFF Zigbee 3.0 USB Dongle is that Zigbee Bridge Pro uses `DIO_8` instead of `DIO_15` to enter BSL bootloader mode. I also spotted difference in `MODE_CONF_0` parameters but I don't know the impact.

So I patched the original HEX file and changed the parameters at 0x57FA8 from:
```
:020000040005F5
:207FA800000080011000C0FFFDFF58003AC1B9FDFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF74
:207FC800FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC50FFEC5FFFFFFFF00FFFFFF00C5C5FF90
:187FE800000000FF00000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF92
:00000001FF
```

to (with fixed checksums accordingly)
```
:020000040005F5
:207FA800000080011000C0FFFDFF58003AC1F9F1FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF40
:207FC800FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC508FEC5FFFFFFFF00FFFFFF00C5C5FF97
:187FE800000000FF00000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF92
:00000001FF
```

Is it possible to have an official firmware with those parameters?

Full story here: https://github.com/arendst/Tasmota/discussions/14419#discussioncomment-2795228

Koenkk, 2022-05-21:
Would it be possible to create a script which converters `CC1352P2_CC2652P_launchpad_coordinator_20220219.hex` to a sonoff zigbee bridge compatible firmware?

s-hadinger, 2022-05-21:
Sure, but it's one more step and error-prone. The main danger is that if someone forgets this extra step and flashes the original firmware with `DIO_15` then BSL mode becomes impossible (and would require JTAG flashing which is way more complicated).

Koenkk, 2022-05-21:
If you can create a PR with a script (Python 3) that would be acceptable for me. If you can place it here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0 I will run it on every release.

Hedda, 2022-05-21:
Please consider building a matching Zigbee Router firmware for this ZBridge-P device as well.

s-hadinger, 2022-05-22:
@Hedda answered in other thread but duplicating here for completeness.

It will be hard to make a router firmware on this device because there is no button connected to the CC2652P and only DIO_8 can be controlled from ESP32, while it is already used for BSL mode.

However, ZNP stack can easily be configured in router mode and this is already supported in Tasmota using a fake `0xFFFF` PanID flag.

Hedda, 2022-05-22:
Yes I understand that a similar problem was solved on the previous ITead Sonoff ZBBridge with EFR32MG21 by using a different Zigbee firmware and most interestingly that supported using the external button on the bridge to put it into pairig/joining mode

https://github.com/arendst/Tasmota/discussions/11536

[https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html](https://www.digiblur.com/2022/02/convert-sonoff-zigbee-bridge-to-router-repeater.html?m=1)

https://community.home-assistant.io/t/inofficial-zigbee-router-firmware-released-for-itead-sonoff-zbbridge-and-zb-gw03-zigbee-gateways-bridges-as-alternative-when-used-with-tasmota-and-or-esphome/393733

laurentppol, 2022-05-22:
> Please make sure that this script will also work with future versions of the stack.
Hmm, how a patch to HEX file could "work with future versions"? Only change in source code will.

s-hadinger, 2022-05-23:
I will prepare a simple script to replace whatever is after `:020000040005F5`. This section is independent from the software version so it should work with future versions.

closed 2022-06-30 by github-actions[bot]

allansari, 2023-03-24:
Did any one manage to hack zigbee bridge pro as router?

s-hadinger, 2023-03-24:
Please see here https://github.com/arendst/Tasmota/discussions/16957

allansari, 2023-03-25:
> Please see here https://github.com/arendst/Tasmota/discussions/16957

https://github.com/arendst/Tasmota/discussions/16957#discussioncomment-5313722