Koenkk/Z-Stack-firmware #375: Increase routing table sizes

Koenkk, 2022-05-27:
To decrease the chances of a `NWK_TABLE_FULL` error (e.g. https://github.com/Koenkk/zigbee2mqtt/issues/12439) the routing table size should be increased (`MAX_RTG_SRC_ENTRIES` and `MAX_RTG_ENTRIES`). Due to a bug in the SDK these tables cannot be increased beyond 255 currently: https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1103762/simplelink-cc13x2-26x2-sdk-cannot-set-max_rtg_src_entries-higher-than-255

closed 2022-07-04 by github-actions[bot]

tteck, 2022-07-25:
Did the new SDK fix this issue?

Koenkk, 2022-07-26:
It seems it is, thanks for pinging, increased the MAX_RTG_SRC_ENTRIES to 500

@ Devon1984 can you check this fw on your network? [`20220726`](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)

tteck, 2022-07-26:
So, this should support larger networks more efficiently, without recurring route discoveries?

Koenkk, 2022-07-26:
@tteck yes, it can cache more source routes

qvntra, 2022-07-29:
I am getting these errors on our new device. I am really far away from the coordinator with no routers when this happens. I'll flash and report back to see if this changes things over the next week.

Devon1984, 2022-09-26:
@Koenkk sorry when you tagged me it didnt send me a ping. I think its because of the space between the @ and my name :-) 
I have been checking up on this one every week but missed this post :-( anyway, im excited to test this new firmware!! Im testing it now and will let you know how it goes.

Devon1984, 2022-09-26:
@Koenkk Its been running now for about an hour and no NWK_TABLE_FULL errors! :-) normally i would have some by now.
Ill let it run for the rest of the day and ill report back here later today on how its performing.

Devon1984, 2022-09-27:
@Koenkk Im letting the FW run for a bit longer to get some more data points but comparing the logs i can see a huge improvement. Previously i would get around 22 NWK_TABLE_FULL errors and now im getitng only 1 or 2 of those errors on the new FW.
I do want it to run for a few more days for some more data comparison though but looking good so far.

Would you like me to send any logs for you to look over?

Devon1984, 2022-09-28:
@Koenkk I have been running this FW for the last 2 days and I have only had 8 instances of NWK_TABLE_FULL errors.
Id say that this is a significant improvement and all my IKEA lights are way more responsive and i have not noticed any device dropouts or relayed reactions so id definetly ask that this change be included in your next master release.

Out of curiosity, is 500 source routes the maximum number or can we go higher? whats the disadvantage of going higher? Im wondering what it would take to completely eliminate the NWK_TABLE_FULL errors.

Koenkk, 2022-09-28:
Great to hear. I've recently improved the firmware further for larger networks. Can you check if this firmware gives more improvements? https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Devon1984, 2022-09-28:
Thats excellent news!! Thanks for working on this and helping us with large networks @Koenkk :-)
As soon as i get home from work today ill load up the FW and report back.

Devon1984, 2022-09-30:
[29th to 30th Sep 22 log.txt](https://github.com/Koenkk/Z-Stack-firmware/files/9683883/29th.to.30th.Sep.22.log.txt)

@Koenkk This FW is most definetly better than the last one! In short, i have had this running now for over 28 hours and i have had NO occurances of the NWK_TABLE_FULL errors which is a GREAT result! I havent noticed any device drop off's either which is amazing.

I have attached my log for information, please start reading from the following time "2022-09-29 15:48:01" before this time i was turning on all my lights and getting the network to settle after the FW upgrade so these errors can be ignored.

You will see a lot of the following errors "(Data request failed with error: 'MAC no ack' (233))" but i believe this is normal because we do use the wall switches in the house to turn off the lights from time to time and of course then the coordinator cant ping the light and hence the error but the most important thing is once i do turn on the light switch there are NO "NWK_TABLE_FULL errors" at all, where before i would be getting them regularly when i turn the lights on.

The only other new error is the following "(Data request failed with error: 'MAC channel access failure' (225))" I have never seen this error before so not sure what this one is but they dont happen very often and i suspect it has something to do with the light switch being off.

Do you know what the difference between these two MAC errors are?

Ill let this FW run now for the rest of the weekend and give it a good workout :-) and give you an update on Monday.

If you have any more FW versions that you would like me to test please pass them on to me and ill load it up and test it for you! especially if you have further improvements for the larger networks :-)

Devon1984, 2022-09-30:
Thanks Koen.
All good then as I do use a 2m USB cable and have the coordinator sitting high up on the  wall away form any other equipment, including the PC that its connected too. 

My coordinator is on zigbee channel 11 and all my local wifi network is on channell 11, 12 and 13. I will confirm the neighbours wifi but im not too concerned as this error doesn't happen very often at all. I was just curious what it actually meant. :-)

Devon1984, 2022-10-03:
@Koenkk I have been running the new FW for the last 3 days and the only day that had any NWK_TABLE_FULL errors in the logs was today. It was only 14 events and it all happened this morning around the same time, but im putting it down to my server as it was busy with a Parity check.

I have been very impressed with the modifications that you have made, they are making a HUGE difference on my side, especially with all the lights in the kids rooms etc. as sometimes the kids turn the switch off when they go to bed. Before this FW i used to get lots of NWK_TABLE_FULL errors and delayed responses when lights were turned off and on at the wall but now im not seeing any of this behaviour.

Definetly one of the best FW releases i have had! Thanks so much!!

Koenkk, 2023-04-01:
Follow up: https://github.com/Koenkk/Z-Stack-firmware/issues/439

Venom84, 2023-04-02:
Thanks for the tag @Koenkk ill get this installed ASAP today and report back.