Koenkk/Z-Stack-firmware #376: Lost control of all zigbee devices after updating to CC1352P2_CC2652P_launchpad_coordinator_20220507 (develop)

alexruffell, 2022-05-29:
``Ever since I updated my Home Assistant system to use a Sonoff USB #.0 Zigbee Dongle Plus running Z-Stack_3.x.0 20220219, I kept having devices drop off daily. It appears that for some at least the dropping off the network was temporary. Also, when I quickly turn on and off a device from HA, the physical device will repeat the on / off cycle several times as if the coordinator keeps sending the command.

Exasperated by this issue, I attempted to flash your latest development firmware (CC1352P2_CC2652P_launchpad_coordinator_20220507.zip) to see if it would help but only ended up with a completely non functional zigbee network. While it appears the coordinator can still find devices (I tried adding a spare sensor), it does not appear to be able to control any of the existing devices.

A random snippet of the logs (too much to paste complete logs) follows:

````
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=136)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=132)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 4 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=137)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=138)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=139)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=140)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.IEEE: 3>, address=08:6b:d7:ff:fe:6e:4a:47), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=129, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x81\x00\x00\x00')
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>), got AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)), retry attempt 5 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=141)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=142)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.IEEE: 3>, address=28:6d:97:00:01:03:dc:2f), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=130, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x82\x00\x00\x00')
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>), got AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)), retry attempt 5 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=143)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.IEEE: 3>, address=ec:1b:bd:ff:fe:fb:15:8c), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=1794, TSN=131, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x83\x00\x00\x04\x00\x00\x00\x02')
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>), got AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)), retry attempt 5 of 5
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.NWK_NO_ROUTE: 205>, Endpoint=1, TSN=144)
2022-05-29 15:56:22 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 1 of 5
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.IEEE: 3>, address=08:6b:d7:ff:fe:6e:5c:c4), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=132, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x84\x00\x00\x00')
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Expected SRSP response AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>), got AF.DataRequestExt.Rsp(Status=<Status.NWK_INVALID_REQUEST: 194>)), retry attempt 5 of 5
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.NWK_NO_ROUTE: 205>), retry attempt 3 of 5
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.zigbee.application] Previously delayed request is now running, delayed by 43.50 seconds
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x079E), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=2820, TSN=146, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x00\x92\x00\x0B\x05\x0D\x05\x0F\x05\x08\x05\x0A\x05')
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.zigbee.application] Previously delayed request is now running, delayed by 43.54 seconds
2022-05-29 15:56:23 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9CEE), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=147, Options=<TransmitOptions.SUPPRESS [... cut ...]
I downgraded to the latest released version but am unable to restore functionality. I will probably upgrade back to 20220507 and attempt to repair all my devices. I still have hopes that the newer firmware will help mitigate my issues with devices dropping off.

Prior to adopting the sonoff coordinator, I was using a Conbee II and did not have any of these issues. I moved to the sonoff as I read it was better and did not suffer from an issue where the network would meltdown randomly after reboots and/or upgrades which happened to me one too many times.

Any chance my issue(s) are related to Z-Stack?

liads, 2022-05-29:
I had a similar issue after the upgrade, but a simple restart of all routers solved it for me.

alexruffell, 2022-05-30:
@liads Thank you for the tip. Unfortunately by the time I saw it, I had already excluded and re-added many routers so the sensors were now orphaned. Thankfully, by not deleting the devices from HA first, adding them back was a breeze. I am not sure whether it is due to 20220507 but adding the sensors back was way better than with 20220219. I did not make any changes and the devices are the same so either I unknowingly cleared some issue with an errant device or the new beta firmware improved something.

Cossey, 2022-06-27:
> ``Ever since I updated my Home Assistant system to use a Sonoff USB #.0 Zigbee Dongle Plus running Z-Stack_3.x.0 20220219, I kept having devices drop off daily. It appears that for some at least the [...]

Could the above be related to [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/12887)?

closed 2022-08-05 by github-actions[bot]