Koenkk/Z-Stack-firmware #378: Compile the firmware using a Mac

tiagomaricate, 2022-06-10:
Hi people.

I just note all the instructions and "official" ways to compile the firmware are using the Flash Programmer 2, only available for Windows.

So, I'd like to know how to do this using a Mac. I can't find any similar tool to do it.

Really sorry if here is not the right place for asking this kind of help. I'm new to using HA and home automation and this is my first time setting up the device. Like a child holding a toy trying to find where to turn it on.

Koenkk, 2022-06-11:
Flash Programer 2 works only on Windows indeed but all the rest works on Mac. I've actually made these instructions while compiling it on a Mac. For flashing on mac use cc2538-bsl or the zigstar tool: https://www.zigbee2mqtt.io/guide/adapters/#notes

tiagomaricate, 2022-06-11:
> Flash Programer 2 works only on Windows indeed but all the rest works on Mac. I've actually made these instructions while compiling it on a Mac. For flashing on mac use cc2538-bsl or the zigstar tool: https://www.zigbee2mqtt.io/guide/adapters/#notes

Perfect! Thanks!

closed 2022-06-11 by tiagomaricate