Koenkk/Z-Stack-firmware #380: Router compile instructions are not working.

jairbj, 2022-07-17:
Hi, I'm trying to compile a router firmware with custom "TX POWER" but compile instructions doesn't seem to work.

1st - When installing [SIMPLELINK-CC13XX-CC26XX-SDK_5.30.01.01](https://www.ti.com/tool/download/SIMPLELINK-CC13XX-CC26XX-SDK) it creates a "simplelink_cc13xx_cc26xx_sdk_5_30_01_01" dir, not "simplelink_cc13x2_26x2_sdk_5_30_01_01" as stated in step 2.

2nd - If I import the projects of step 3, there are no files "znp.syscfg" inside projects. I only found "zr_genericapp.syscfg".

3rd - If I open the file I mentioned on previous step, I didn't find the "power management" section.

4th - If I try to apply the diff to the workspace directory I get:
```
/mnt/c/ti/workspace_v11 ❯ git apply firmware.patch --ignore-space-change                              
firmware.patch:160: trailing whitespace.
--preinclude=preinclude.h
firmware.patch:807: trailing whitespace.
    // Set TXRX pin to 0 in RX and high impedance in idle/TX.
firmware.patch:827: trailing whitespace.
    // Set TXRX pin to 0 in RX/TX and high impedance in idle.
firmware.patch:1283: trailing whitespace.
--preinclude=preinclude.h
firmware.patch:1504: trailing whitespace.
--preinclude=preinclude.h
warning: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/Application/zcl_genericapp.c has type 100755, expected 100644
warning: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/Application/zcl_genericapp_data.c has type 100755, expected 100644
warning: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/Stack/Config/f8wrouter.opts has type 100755, expected 100644
warning: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/app.cfg has type 100755, expected 100644
warning: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg has type 100755, expected 100644
error: patch failed: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg:50
error: zr_genericapp_CC1352P_2_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg: patch does not apply
warning: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/Application/zcl_genericapp.c has type 100755, expected 100644
warning: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/Application/zcl_genericapp_data.c has type 100755, expected 100644
warning: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/Stack/Config/f8wrouter.opts has type 100755, expected 100644
warning: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/app.cfg has type 100755, expected 100644
warning: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg has type 100755, expected 100644
error: patch failed: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg:47
error: zr_genericapp_CC26X2R1_LAUNCHXL_tirtos_ccs/zr_genericapp.syscfg: patch does not apply
warning: zr_genericapp_LP_CC2652RB_tirtos_ccs/Application/zcl_genericapp.c has type 100755, expected 100644
warning: zr_genericapp_LP_CC2652RB_tirtos_ccs/Application/zcl_genericapp_data.c has type 100755, expected 100644
warning: zr_genericapp_LP_CC2652RB_tirtos_ccs/Stack/Config/f8wrouter.opts has type 100755, expected 100644
warning: zr_genericapp_LP_CC2652RB_tirtos_ccs/app.cfg has type 100755, expected 100644
warning: zr_genericapp_LP_CC2652RB_tirtos_ccs/zr_genericapp.syscfg has type 100755, expected 100644
error: patch failed: zr_genericapp_LP_CC2652RB_tirtos_ccs/zr_genericapp.syscfg:48
error: zr_genericapp_LP_CC2652RB_tirtos_ccs/zr_genericapp.syscfg: patch does not apply
```

Koenkk, 2022-07-17:
I will update the instructions with the next TI SDK release (which I expect soon)

jairbj, 2022-07-17:
Just for known, why not release the full source of the firmware?

Koenkk, 2022-07-18:
The project file contains all types of computer specific paths so it will not be usefull.

bojanpotocnik, 2022-07-19:
I thought this is actually pre-compiled/prepared FW using https://github.com/ptvoinfo/zigbee-configurable-firmware - or are there any special modifications?

closed 2022-08-26 by github-actions[bot]