Koenkk/Z-Stack-firmware #381: Please have old firmware versions available

gzcwnk, 2022-07-19:
I have flashed my sonoff3.0 as a router but it wont join, I have raised a bug request.   I cant prove its flashed but if I had an older firmware version I could flash it and check the logs and re-flash it and check the logs, thus proving its indeed has been flashed.

gzcwnk, 2022-07-19:
delete this please or close it.  I just dont know how to use this web page, I found the older firmware TY.

gzcwnk, 2022-07-19:
my error....

gzcwnk, 2022-07-19:
Actually that's the co-ordinator firmware, I need older router firmware please.

gzcwnk, 2022-07-19:
please put up a URL, TY

bojanpotocnik, 2022-07-26:
https://github.com/Koenkk/Z-Stack-firmware/commits/master/router on the right side there is a _View at this point in the history_ button.

closed 2022-09-02 by github-actions[bot]