Koenkk/Z-Stack-firmware #384: How to Debug Z-Stack_Home_1.2

King-anderson, 2022-08-03:
Hi @Koenkk 
I'm reviewing Z-Stack_Home_1.2 stack source code to find out why it crashes (possibly a memory leak issue)
I want to debug in online mode (using IAR and CCDEBUGGER).
but if I want to use Debug project instead of ProdHex, the linker says flash memory capacity is not enough. it's strange. I can't use Debug project at all. even if I use the default configuration.
in ProdHex project, I should set the optimization parameter on medium or High which the debugger can't start with main.c code lines! 

summarize: I can't debug z-stack source code lines using cc-debugger.

have you ever debug it using a debugger? do you know any way to see any debug logs (z-stack log)?

Koenkk, 2022-08-03:
@King-anderson I've never debugged a CC2530/CC2531 with the ccdebugger. I suggest asking this on the e2e TI forum.

Koenkk, 2022-08-10:
> but adapter doesn't respond to them. is "route request" command handled in zigbee-herdsman or z-stack?

Route requests are handled in z-stack. However the route requests are not send to the coordinator specifically but are a broadcast instead (all devices will receive it when destination `0xFFFF`). If no device has the route nobody will respond. To who is the route requested? (do you get more details when clicking on the "route request"?)

King-anderson, 2022-08-29:
Hi @Koenkk 
sorry for the delay. to answer your question I had to repeat my test which is time-consuming. 
I join 19 pcs routers to my cc2530+z2m hub. everything was ok. I turned off 13 routers and then turned on them simultaneously. the routers started to send "route request" or "Network address request" to the adaptor simultaneously. after these requests adapter crashed. every time I perform this test, the adapter is crashed. but when I reset the adapter by lowing its pin and restarting the z2m, it works fine. 
I guess, the crash problem relates to handling multiple routing requests and may be resolved by changing the z-stack preinclude.h file macro values.
below is the network log in normal and crashed conditions:

1- normal condition:
[image]

2- first time after turning off and on 13 pcs routers simultaneously:
[image]

3- I restart adapter and z2m and repeat the test (step 2):
[image]

Koenkk, 2022-08-29:
> may be resolved by changing the z-stack preinclude.h file macro values.

I guess you are talking about the routing table sizes. These are the number of routes that can be stored at the coordinator side, it doesn't have anything to do with responding to them (which is the case here).

zen2, 2022-08-31:
I follow the discussion from far since I'm not impacted with this issue.

I wonder the firmware use some algorithms that depend on routing table size.
If it's the case: depending on algorithms complexity the routing table size can involve:
- timing issue: algorithms treatment take longer because of bigger routing table size (but no crash)
- memory issue: algorithms treatment need more memory because of bigger routing table size and so can crash the device in case of no enough memory to complete the treatment

I've always suspect this kind of problem with old CC2531 coordinator when using a lot of router devices with extended routing table size.

It's just my 2 cents idea.

King-anderson, 2022-09-03:
Hi @Koenkk 
can you tell me the "Defined symbols" you used for [CC2530_SOURCE_ROUTING_20211116](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2530_SOURCE_ROUTING_20211116.zip)

Koenkk, 2022-09-06:
@King-anderson non except the ones mentioned in step 6: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md

closed 2022-10-14 by github-actions[bot]