Koenkk/Z-Stack-firmware #389: Reduce the APS ack wait duration

pipiche38, 2022-08-31:
here https://github.com/Koenkk/Z-Stack-firmware/blob/3406c6a0cd2b755f302f4cacd07b37497a86cc08/coordinator/Z-Stack_3.x.0/firmware.patch#L350 it looks like the APS Ack timeout as been reduced from 6s to 1s. What was driving this request ? 

The question is related to ZED like eTRV which are configured on a 6s polling interval (default Zigbee). In such cases we will have a high probablility to get Ack timeout when send a new Setpoint to an eTRV type of devices !

Koenkk, 2022-08-31:
This makes the recovery mechanism kick in faster (route discovery/network address retrieval). Could you check if settings this back to 6s indeed fixes your issue?

pipiche38, 2022-08-31:
This was more a question rather than an existing issue.
I've integrated a number of eTRV Tuya and Danfoss/Popp and I know they rely on the polling mecanism (which is done on a 6 secondes window), so I'm trying to understand the impact of the 6 to 1 .

WIll see If I can set a test bed for that. 

Side question:

Is z2m rely on the 6 seconds, or do you do retry when getting Ack ?
I know that zigpy-znp is doing 5 retries with different ways, which for me doesn't make sense, as we should retry on the firmware stack for that which is doing the proper retry during the 6 seconds timeout windows

Koenkk, 2022-08-31:
> Is z2m rely on the 6 seconds, or do you do retry when getting Ack ?

z2m doesn't do this, this is handled in the stack. But z2m will retry requests when it fails (route discovery, network address retrieval)

closed 2022-11-18 by github-actions[bot]