Koenkk/Z-Stack-firmware #390: Compile Errors for Z-Stack 3.x.0

bulldog5046, 2022-09-01:
There seem to be some errors when following the compile instructions.

Firstly, applying the git patch file:

```./firmware.patch:284: trailing whitespace.

./firmware.patch:286: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since
./firmware.patch:1240: trailing whitespace.
    // Set TXRX pin to 0 in RX and high impedance in idle/TX.
./firmware.patch:1258: trailing whitespace.
    // Set TXRX pin to 0 in RX/TX and high impedance in idle.
warning: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
./firmware.patch:1459: new blank line at EOF.
+
error: patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg:128
error: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply
warning: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
error: patch failed: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/znp.syscfg:121
error: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply
warning: znp_LP_CC2652RB_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
error: patch failed: znp_LP_CC2652RB_tirtos_ccs/znp.syscfg:122
error: znp_LP_CC2652RB_tirtos_ccs/znp.syscfg: patch does not apply
```

And secondly, Step 8 which reads 

> Build the 3 projects; right click -> Build project.
>Important: by default the launchpad variant of the CC1352P2_CC2652P (= znp_CC1352P_2_LAUNCHXL_tirtos_ccs) is build. To build the other variant comment #define LAUNCHPAD_CONFIG 1 in preinclude.h (located under Stack/Config/), don't forget to save.

preinclude.h does not exist.

I'm not sure the latter should really be a problem in my case but the built firmware does not function correctly. It flashes, and seems to respond to some commands but the NVRam doesn't look right when backed up and it fails to form a network. In contract the precompiled firmware works fine.

I could use some help as i need to change the default TC link key and this is the only was to do it.

bulldog5046, 2022-09-01:
Further. If you manually correct the changes to znp.syscfg

```scripting.excludeFromBuild("ti_devices_config.c");
scripting.excludeFromBuild("ti_radio_config.c");
scripting.excludeFromBuild("ti_radio_config.h");
scripting.excludeFromBuild("ti_drivers_config.h");
```
This causes a build failure due to missing modules.

Removing these lines results in build failure due to size issues.

```error #10324-D: BOUND section ".TI.bound:flashBuf0" spans the boundary of an existing memory range FLASH
error #10099-D: program will not fit into available memory, or the section contains a call site that requires a trampoline that can't be generated for this section. run placement with alignment fails for section ".TI.bound:flashBuf0" size 0x6000
error #10010: errors encountered during linking; "znp_CC1352P_2_LAUNCHXL_tirtos_ccs.out" not built
 
>> Compilation failure
makefile:265: recipe for target 'znp_CC1352P_2_LAUNCHXL_tirtos_ccs.out' failed
gmake[1]: *** [znp_CC1352P_2_LAUNCHXL_tirtos_ccs.out] Error 1
makefile:261: recipe for target 'all' failed
gmake: *** [all] Error 2

**** Build Finished ****
```

Koenkk, 2022-09-05:
Can you try the instructions of the new SDK? https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/COMPILE.md

Regarding the `znp.syscfg` make sure you resave those files as explained in step 5.

bulldog5046, 2022-09-06:
After digging through for awhile I eventually discovered the --reject switch for git apply. I think it's windows file permissions not playing well with git.