Koenkk/Z-Stack-firmware #392: onoff ZBDongle-P

parvbarot, 2022-09-29:
I've flashed my Sonoff ZBDongle-P with [CC1352P2 CC2652P launchpad router 20220125.zip] firmware. I wanted to add this to ZHA integration, but it wasn't picked up. I had flashed it numerous times, but ZHA wasn't picking it up.

Xploder, 2022-10-01:
You've flashed the router firmware, which just works for "Repeater-Type" functionality.
You'll need to flash the coordinator firmware.

parvbarot, 2022-10-02:
I am using sonoff zigbee bridge as coordinator and wants to use zigbee usb dongle 3.0 (CC2652P) as Router but after flashing ZHA didn't detect it.

parvbarot, 2022-10-02:
I tried using Zigbee2Mqtt, but it didn't detect. But as a coordinator, this dongle functions perfectly.

jerrm, 2022-10-04:
The device doesn't stay in pairing mode very long.  

Unplug the stick, place ZHA into pairing mode, then plug the stick back into power.  It should pair within a few seconds.

closed 2022-11-12 by github-actions[bot]