Koenkk/Z-Stack-firmware #393: Sonoff Dongle-P - Unable to PING

keness521, 2022-10-02:
I hope this is in the correct place, I apologize if not!

I flashed a factory fresh Sonoff Dongle-P with the v20220908 ROUTER firmware, although my first pairing attempt failed (interview failed) the second went through fine.

I connected it directly to my coordinator (also a Sonoff Dongle-P, v20220219 firmware) which is quite distant, but it still shows 57/65 lqi to the coordinator in the Z2M map.

It worked fine once the interview completed, and immediate took on a million dashed lines in the Z2M map, However, after just a few minutes, the PING will fail (timeout) twice and it will go OFFLINE. I can reconfigure, it will come back, but within just a few minutes, it will do the same.

Is it too far from the coordinator, even though the lqi looked reasonable? I forced it to join the coordinator, should I just "Permit Join (All)" and let it do what it wants instead? Or am I using a non-primetime version of the firmware perhaps?

I've heard good things about these as routers. And my coordinators using them are great, minus one with defect which caused exceptionally poor reception... Long since trashed that one. So I have to think it is my fault or situation, not the device/firmware.

Thanks for any help you might be able to provide.

keness521, 2022-10-03:
Quick followup, I flashed a different Dongle-P with v20220125 ROUTER firmware, and joined it to a completely separate network. In this case, it is only about 4 feet away from the coordinator of that network.

Exact same results... After a couple minutes, I get a failed to PING message, then shortly later, another, and after two failures, it goes "Disabled".

I must be doing something wrong...?

keness521, 2022-10-03:
I'm closing the comment here, because I think I am in the wrong place. Apologies!

closed 2022-10-03 by keness521

Fyase82, 2023-11-29:
I have the exact same issue. SonOff dongle P with the latest router firmware do not respond to ping but it seems to work anyway.... Do you need some logs?