Koenkk/Z-Stack-firmware #396: Error when applying the firmware.patch

pipiche38, 2022-10-19:
Don't know if that is important or not, but while applying

`workspace % git apply firmware.patch --ignore-space-change`

I got the following errors

```

firmware.patch:281: trailing whitespace.

firmware.patch:283: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since
firmware.patch:1231: trailing whitespace.
    // Set TXRX pin to 0 in RX and high impedance in idle/TX.
firmware.patch:1249: trailing whitespace.
    // Set TXRX pin to 0 in RX/TX and high impedance in idle.
firmware.patch:1450: new blank line at EOF.
+
error: patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ticlang/znp.syscfg:125
error: znp_CC1352P_2_LAUNCHXL_tirtos_ticlang/znp.syscfg: patch does not apply
```

Plateform: Mac Os
Dev Stack: simplelink_cc13xx_cc26xx_sdk_6_20_00_29/examples/rtos

Koenkk, 2022-10-20:
Did you do step 5 of https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/COMPILE.md ?

pipiche38, 2022-10-20:
I think so.
Will redo tonight

pipiche38, 2022-10-20:
You are probably right I didn't do it.
Thanks for taking the time to respond

closed 2022-10-20 by pipiche38

Koenkk, 2022-10-21:
No problem, if I may ask, what is your goal of the firmware compilation? If you find any possible improvements please let me know.

pipiche38, 2022-10-21:
Sure. If you recall I asked you about the APS ACK timeout value which is 1s on your firmware, and which is 6 seconds on Zigate.
I discover that I'm having issue with battery devices when doing OTA on ZNP based firmware, but nothing on the Zigate and nothing on bellows (silicon labs).
So I wanted to do a test while putting the firmware with a timeout of 6s and see how it behaves.

For instance I don't think that making retrys, route-discovery makes a lot of sense when you are just waiting for an end device doing it's regular polling window.
I understand on this other side, that for main powered device, the approach to reduce the APS ACK to 1s has a lot of benefit

mbenfe, 2022-12-10:
Hi tried as weel. same sdk and ccs releases, all steps and still have issues:
firmware.patch:1459: new blank line at EOF.
+
error: patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg:4
error: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply
warning: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755
warning: znp_LP_CC2652RB_tirtos_ccs/cc13x2_cc26x2.cmd has type 100644, expected 100755

i turned things in all directions ... redo and redo but still the same :-)

Sheng2216, 2023-04-10:
> Hi tried as weel. same sdk and ccs releases, all steps and still have issues: firmware.patch:1459: new blank line at EOF. + error: patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg:4 error: [...]

I guess you are working on a Windows machine, 755 means that the directory has the default permissions -rwxr-xr-x (represented in octal notation as 0755), while 644 means owner can read/write, group/others can read only. I had the same issue and I find out that Windows break the permission sometimes so I switched to a Linux virtual machine, the warning about 100644 is gone, but I still have the error: `patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ticlang/znp.syscfg:4` error, hmm...

mbenfe, 2023-04-10:
Used code composer patch system right click project root->team->apply patch and the it worked nicely. 
Thanks

cghui2019, 2024-07-29:
I have same error，Has anyone solved this problem?

Plateform: windows
Dev Stack: simplelink_cc13xx_cc26xx_sdk_7_10_00_98/examples/rtos 

warning: source/ti/zstack/boards/cc13x2_cc26x2/cc13x2_cc26x2_tirtos7_ticlang.cmd has type 100644, expected 100755
warning: source/ti/zstack/boards/cc13x2x7_cc26x2x7/cc13x2x7_cc26x2x7_tirtos7_ticlang.cmd has type 100644, expected 100755
error: patch failed: workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg:4
error: workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg: patch does not apply

cghui2019, 2024-07-29:
> Don't know if that is important or not, but while applying
>
> `workspace % git apply firmware.patch --ignore-space-change`
>
> I got the following errors
>
> ```
>
> firmware.patch:281: trailing whitespace. [...]

Did you manage to solve this problem in the end? I'm facing the same issue now