Koenkk/Z-Stack-firmware #397: Error when flashing the new firmware

minimmi, 2022-11-04:
Hello,

I am currently trying to flash the latest router firmware: 

https://github.com/Koenkk/Z-Stack-firmware/blob/develop/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20221102.zip

but I always get the following error message: 

[image]

What am I doing wrong? Can anyone help me?

danieledwardgeorgehitchcock, 2022-11-04:
I'm getting the same.. A bit of googling suggests it may be something to do with the method that was used to compile the hex..?

tube0013, 2022-11-04:
did you unzip the file?

danieledwardgeorgehitchcock, 2022-11-04:
Yes and followed the flashing guidelines. Haven't had an issue with previous firmwares.. just this one

minimmi, 2022-11-04:
> Yes and followed the flashing guidelines. Haven't had an issue with previous firmwares.. just this one

I can confirm that. With the previous firmware version everything works without problems. If I then try to flash the new in the same way, I get the error message.

kfraleigh, 2022-11-04:
Same error here as well using the TI software to update the Sonoff Zigbee 3.0 dongle.

boris1000, 2022-11-05:
Same here for a Sonoff CC2652P

>Initiate access to target: COM6 using 2-pin cJTAG.
>Reading file: C:/ATemp/ZIGBEE Firmware/Coordinator/CC1352P2_CC2652P_launchpad_coordinator_20221102.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.
>Initiate access to target: COM6 using 2-pin cJTAG.
>Reading file: C:/ATemp/ZIGBEE Firmware/Coordinator/CC1352P2_CC2652P_launchpad_coordinator_20221102.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.
>Initiate access to target: COM6 using 2-pin cJTAG.
>No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Connecting over serial bootloader failed: No response from device. Device may not be in bootloader mode. Reset device and try again.
If problem persists, check connection and baud rate.
>Failed to create device object.

boris1000, 2022-11-05:
used  now the stable one which works without an issue , using Texas Instruments flasher

boris1000, 2022-11-05:
Just to add - this the same error when trying the router firmware

jchurchward, 2022-11-05:
Same here, unable to flash as per what others are saying above.

alexruffell, 2022-11-06:
Follow these instructions

https://github.com/Koenkk/Z-Stack-firmware/issues/383#issuecomment-1301903658

Koenkk, 2022-11-06:
Reported this to TI: https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1168418/simplelink-cc13x2-26x2-sdk-flashing-simlelink-6_30_00_84-hex-through-uniflash-fails

Koenkk, 2022-11-06:
Seems that the issue is specific to Uniflash 8.1.0, Uniflash 8.0.0 works fine for me.

minimmi, 2022-11-06:
Downloaded UniFlash here:
https://www.ti.com/tool/download/UNIFLASH/8.0.0

I tried to flash the firmware with Uniflash, but I can't get it to work because my dongle is not recognized: 

[image]

kfraleigh, 2022-11-06:
TI Flash Programmer 2 is what I have been using (and others posting) that doesn’t work with the new firmware. I hadn’t seen Uniflash mentioned for flashing the Sonoff dongle.

minimmi, 2022-11-06:
Update:

Have now tried to flash the firmware with the help of the python script as described in this video:

https://youtu.be/iCE5Z43EKpk?t=1254

It looked like everything worked out:

[image]

but after i reconnected the stick, zigbee2mqtt still shows me the old firmware:

[image]

mikecay, 2022-11-06:
I'm getting the same error that others have reported.

>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: Z:/Projects/Zigbee_Flash/CC1352P2_CC2652P_launchpad_router_20221102/CC1352P2_CC2652P_launchpad_router_20221102.hex.
>Unknown record type: 3.
>Reset target ...

I can still flash the older FW but not this new one.

Koenkk, 2022-11-07:
TI responded that this is a known issue of Uniflash 8.1.0, so nothing we can do from the z2m side. Use Uniflash 8.0.0 or cc2538-bsl or zigstar flasher instead.

https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1168418/simplelink-cc13x2-26x2-sdk-flashing-simlelink-6_30_00_84-hex-through-uniflash-fails

minimmi, 2022-11-07:
> TI responded that this is a known issue of Uniflash 8.1.0, so nothing we can do from the z2m side. Use Uniflash 8.0.0 or cc2538-bsl or zigstar flasher instead. [...]

As you can see in my last comment, I did the flashing with cc2538-bsl. In zigbee2mqtt I still see the old firmware.

Shamshala, 2022-11-07:
@minimmi Try it again, I succesfully flashed coordinator via cc2538-bsl and it works.

BastiaanNaber, 2022-11-07:
Downgrading to Uniflash 8.0.0 worked for me. Thanks!

cpuks, 2022-11-09:
Also please try ZigStar Multitool (https://zig-star.com/radio-docs/zigstar-multi-tool/) developed for zigstar works with any CC2652p adapter.

boris1000, 2022-11-09:
I used now successfully the ZigStar Multitool. Is a bit tricky for writing the firmware.
First it detects the COM Port, when starting writing you need to push the Boot button again multiple times and click on start in parallel. Suddenly it will start writing -  is a bit confusing this behaviour but it worked.

evelant, 2022-11-10:
cc2538-bsl woked for me, used a windows machine, command such as:

```python cc2538-bsl.py -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb .\PATH_TO_FIRMWARE.hex```

epower53, 2022-11-10:
I was also successful with the python script, run from a Win11 machine to update my Sonoff Zigbee Dongle Plus:

`py cc2538-bsl.py -e -w -v -p com3 --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20221102.hex`

[image]

minimmi, 2022-11-10:
I just don't know what I'm doing wrong or what it could be?! I have now flashed it again successfully with both cc2538-bsl.py and ZigStar. But every time the old firmware is displayed afterwards in zigbee2mqtt:

[image]

jchurchward, 2022-11-10:
> I used now successfully the ZigStar Multitool. Is a bit tricky for writing the firmware. First it detects the COM Port, when starting writing you need to push the Boot button again multiple times [...]

Thanks for that. I was able to get mine working as well with some fiddling around.

boris1000, 2022-11-10:
> I just don't know what I'm doing wrong or what it could be?! I have now flashed it again successfully with both cc2538-bsl.py and ZigStar. But every time the old firmware is displayed afterwards in zigbee2mqtt:
> 
> [image]

And you are using the dev version? I never checked afterwards...

minimmi, 2022-11-10:
> > I just don't know what I'm doing wrong or what it could be?! I have now flashed it again successfully with both cc2538-bsl.py and ZigStar. But every time the old firmware is displayed afterwards in zigbee2mqtt:
> > [image]
> 
> And you are using the dev version? I never checked afterwards...

Yes, I use this version:

https://github.com/Koenkk/Z-Stack-firmware/blob/develop/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20221102.zip

I am interested in the possibility of adjusting the transmitting power, but I still do not have this option.

jerrm, 2022-11-10:
> But every time the old firmware is displayed

For Router firmware?

I had the same problem.   It's more a z2m issue than a flashing issue.

 @Koenkk recommended doing a force remove, restart, then re-pair. Worked for me.

minimmi, 2022-11-11:
> > But every time the old firmware is displayed
> 
> For Router firmware?
> 
> I had the same problem. It's more a z2m issue than a flashing issue.
> 
> @Koenkk recommended doing a force remove, restart, then re-pair. Worked for me.

Thanks for this tip. After force remove, restart and re-pair it shows now the correct firmware. 

[image]

I can confirm under these circumstances that flashing with Zigstar as well as with cc2538-bsl.py works. 

But I still have no way to set the transmit power:

[image]

znanev, 2022-11-11:
@minimmi If you update to the latest development code of z2m, the option for changing the transmit power will be there:

[image]

However I don't recomment you to update to the latest dev yet, as there is a bug preventing pairing of new devices:

https://github.com/Koenkk/zigbee2mqtt/issues/14925#issuecomment-1311619501

I've patched the code as a workaround for the time being in order to be able to test the new router firmware, but unfortunately setting the transmit_power does nothing and I get the following errors in z2m's log:

```
Error 2022-11-11 14:42:44Publish 'set' 'transmit_power' to '0x00124b00228123a8' failed: 'Error: Write 0x00124b00228123a8/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 64718 - 242 - 44 - 0 - 4 after 10000ms)'
```

epower53, 2022-11-11:
That error is related to the dev branch of z2m, not the firmware. I see none of the issues described in [z2m issue# 14925](https://github.com/Koenkk/zigbee2mqtt/issues/14925#issuecomment-1311619501) when using the main branch of z2m (1.28.2) and dev firmware 20221102 on a Sonoff Dongle-P.

minimmi, 2022-11-11:
> @minimmi If you update to the latest development code of z2m, the option for changing the transmit power will be there:
> 
> [image]
> 
> However I don't recomment you to update to the latest dev yet, as there is a bug preventing pairing of new devices:
> 
> [Koenkk/zigbee2mqtt#14925 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/14925#issuecomment-1311619501)
> 
> I've patched the code as a workaround for the time being in order to be able to test the new router firmware and setting _transmit_power_ indeed works as intended.

Ah ok, I didn't know that I needed the dev version of z2m for this. Thanks for your answer. Then I'll wait until the dev version is tested and officially released.

nkrepo, 2022-11-25:
> Thanks for this tip. After force remove, restart and re-pair it shows now the correct firmware.
>
> [image]
>
> I can confirm under these circumstances that flashing with Zigstar as well as with [...]

I flashed an already paired sonoff ZBDongle-P with Z-Stack_Home_3.x.0_20220125 router firmware using the cc2538-bsl script. All went smoothly and the firmware was flashed without issues.
As it is already reported the device shows up that is has the previous firmware.

[image]

I am using the latest release of zigbee2mqtt (1.28.2) not from the dev channel that it has the pairing issue. I forced removed the device but afterwards the device was recognized as unknown device. 

@Koenkk 
Is something that I should expect ?

znanev, 2022-11-28:
@nkrepo Shouldn't you be using [CC1352P2_CC2652P_**launchpad**_router_20220125.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20220125.zip) with ZBDongle-P (instead of CC1352P2_CC2652P_**other**_router_20221102.zip) ?

I'm also using a ZBDongle-P and flashed it with the _launchpad_router_ firmware variant, as seen in the last row of the table here:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

dreimer1986, 2022-12-01:
Slight update here. UniFlash 8.1.1 works fine. I read that some used the SmartRF Flash Programmer 2 as I did, too.
Nope. Still v1.8.3 and still unuseable...

minimmi, 2022-12-02:
Short update from my side:

- cc2538-bsl.py works
- ZigStar works 
- TI Flash Programmer v2 does not work
- Uniflash 8.1.X does not work
- Uniflash 8.0.0 works partially ( not for me, because stick is not recognized. For others it seems to work) 

Should this stay open until it works with all methods or can it be closed, what do you say?

toxic0berliner, 2022-12-02:
had also the same issue with TI flash programmer 2 ver. 1.8.2 and was sucessful with cc2538-bsl.py ;)

Dylantje, 2022-12-02:
Is there some that have flash a sonoff plus dongle flashed with the new firmware?

What software is working the best.
And simple?
Noob here.

Thanks guys

znanev, 2022-12-02:
@Dylantje I find the `cc2538-bsl.py` script working flawless, every time. It's simple to use it too. You just need to have Python installed first, then invoke the script like this:

```python cc2538-bsl.py -e -w -v -p PORT firmware.hex```

where **`PORT`** is the serial port where your ZigBee device has been attached.

In Linux that might be **`/dev/ttyUSB0`**, in Windows - **`COM4`** etc.

Dylantje, 2022-12-02:
@znanev 
Thanks 
Only never worked with python. Running on a windows pc..
Will try it, only can i brick the stick?

Prefere a windows option for me.
Perhaps a good working option

jerrm, 2022-12-02:
I haven't used it, but the zigstar tool is open source, cc2538-bsl based, and the releases include both Windows and Mac GUI builds.

https://github.com/xyzroe/ZigStarGW-MT

nkrepo, 2022-12-02:
> nkrepo
Good to notice ,  I have indeed flashed with CC1352P2_CC2652P_launchpad_router_20221102 , it was just a text copy paste error.
I upgraded to latest zigbee2mqtt 1.28.4 and the now the sonoff p dongle flashed with 20221102 is paired without any issue.

Venom84, 2022-12-02:
Hi Guys. im trying to flash my CC2652R zzh coordinator with the new 20221102 firmware but on Uniflash it doesnt auto detect my coordinator and from their list i cant see a CC2652R. I do see a CC2652R1F and a 1F(Bootloader) any advice which one to choose?

minimmi, 2022-12-05:
> @znanev Thanks Only never worked with python. Running on a windows pc.. Will try it, only can i brick the stick?
> 
> Prefere a windows option for me. Perhaps a good working option

Follow the instructions in this video:

https://youtu.be/iCE5Z43EKpk?t=1254

With this I managed to do it without any problem.

Nik71git, 2022-12-08:
Same as yours: problems with TI flash programmer but it works with python script (I mean Sonoff 3.0 and router firmware) and now ZHA (🤫) has power transmitter slider too. Thanks @Koenkk

joe-sydney, 2022-12-10:
I used a Hex2bin converter on the .hex file which converted it to a .bin file and was able to flash that using the TI Flash Programmer.

delinend, 2022-12-14:
I also tryed to convert to BIN, with Hex2bin, and now it works with TI Flash :-)

tszdr, 2023-03-06:
Following procedure worked for me on Sonoff ZIgbee 3.0 USB Dongle Plus (ZBDongle-P):

1. Converting CC1352P2_CC2652P_launchpad_router_20221102.hex to binary format using https://srecord.sourceforge.net/: 
srec_cat.exe "CC1352P2_CC2652P_launchpad_router_20221102.hex" -intel -o CC1352P2_CC2652P_launchpad_router_20221102.bin -binary

2.  Programming CC1352P2_CC2652P_launchpad_router_20221102.bin file using TI SmartRF Flash Programmer 2 (ver. 1.8.2) (target: CC2652P)

smcallis77, 2023-03-15:
> Following procedure worked for me on Sonoff ZIgbee 3.0 USB Dongle Plus (ZBDongle-P):
>
> 1. Converting CC1352P2_CC2652P_launchpad_router_20221102.hex to binary format using [...]

Worked for me.

alikking, 2023-04-02:
> Worked for me.

How to use srecord? I have no idea! Have download it but there is noch app to open and use for converting it.

tszdr, 2023-04-06:
Download srecord from sourceforge, unzip and execute srec_cat from bin folder as per the the command line example above.

devilization, 2023-04-16:
This is what i did:
1.	Install ‘SRecord’
2.	Open power shell with administrator tight
3.	Guide to the ‘SRecord’ install folder ‘C:\Program Files\srecord\bin’
4.	Copy the firmware (CC1352P2_CC2652P_launchpad_router_20221102.hex) to above folder
5.	Run this command to start converting the files
.\srec_cat.exe "CC1352P2_CC2652P_launchpad_router_20221102.hex" -intel -o CC1352P2_CC2652P_launchpad_router_20221102.bin -binary

[image]

alikking, 2023-04-19:
A Big Thanks to You!!!

> This is what i did:
>
> 1. Install ‘SRecord’
> 2. Open power shell with administrator tight
> 3. Guide to the ‘SRecord’ install folder ‘C:\Program Files\srecord\bin’
> 4. Copy the firmware [...]

mendoncart, 2023-05-03:
> This is what i did:
>
> 1. Install ‘SRecord’
> 2. Open power shell with administrator tight
> 3. Guide to the ‘SRecord’ install folder ‘C:\Program Files\srecord\bin’
> 4. Copy the firmware [...]

It worked! Thanks!

[image]

If anyone is crazy enough, [here](https://uploadnow.io/f/Ftjw17B) is the firmware bin file so you can flash with Flash Programmer 2:
(I AM NOT RESPONSIBLE FOR THIS FILE, USE AT YOUR OWN RISK)

sphinxlady, 2023-08-07:
> 5\. .\srec_cat.exe "CC1352P2_CC2652P_launchpad_router_20221102.hex" -intel -o CC1352P2_CC2652P_launchpad_router_20221102.bin -binary

Worked like a charm thanks for the heads up :)

mztimur, 2023-08-08:
> This is what i did:
>
> 1. Install ‘SRecord’
> 2. Open power shell with administrator tight
> 3. Guide to the ‘SRecord’ install folder ‘C:\Program Files\srecord\bin’
> 4. Copy the firmware [...]

CCS already has all the necessary utilities for making a bin file. You can add this line in project settings Project>Properties>Build>Steps>Post-build steps

`"${CCS_INSTALL_ROOT}/utils/tiobj2bin/tiobj2bin" "${BuildArtifactFileName}" "${BuildArtifactFileBaseName}.bin" "${CG_TOOL_ROOT}/bin/tiarmofd" "${CG_TOOL_ROOT}/bin/tiarmhex" "${CCS_INSTALL_ROOT}/utils/tiobj2bin/mkhex4bin"`

As shown in the picture below
[image]

Wickedy123, 2023-08-08:
Just to share my experience:

- SmartRF utility works fine, although it appears necessary to convert to bin format. I used 'hex2bin'

**- The Sonoff stick drops out of boot mode very quickly - within 20 seconds**. So you need to keep plugging it back in whilst holding down boot button for 5 seconds or so. This must be repeated for every attempt

dsduarte, 2023-08-25:
I started by trying to use UniFlash but got confused looking for the right chip to choose for sonoff zigbee dongle plus...
so I did the .hex -> .bin using srecord and everething was OK!

gravelfreeman, 2023-12-15:
I was able to avoid installing Python.

There's a workaround with ZigStarGW on Windows.

- Plug the dongle and hold the button 20 seconds
- Try to flash and get "timeout" message
- Unplug and plug the dongle immediately
- Try to flash again and it'll work

Texas Instrument is crappy.

[19 more replies]