Koenkk/Z-Stack-firmware #403: Error while CC2531 firmware flashing

balebanov, 2022-12-07:
Hello!
I have an RPi 4B with Ubuntu onboard. I've bought 2 ZigBee CC2531 USB-sticks and I want to flash them.

I use this readme: https://jmichault.github.io/flash_cc2531-dok/en/
After connection to RPi's GPIO the command `./cc_chipid` returned this:
`  ID = b524.`
Then I use command `./cc_erase` to erase the memory. Result is:
```
  ID = b524.
  erase result = 00a2.
```
The first question: is the erase result 00a2 normal?

Then I write the hex-file to flash memory with the command: 
```
./cc_write CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page   6/128.
flash error... Have you erased before write ?
```
Number of page where the program stopped (6) is different every time.
What can you advice to avoid this problem?
BR

balebanov, 2022-12-07:
Problem solved :)

closed 2022-12-07 by balebanov

NardoSnyman, 2023-02-28:
@balebanov , how did you resolve this problem? I am facing the same challenge... *Thanks!

balebanov, 2023-02-28:
> @balebanov , how did you resolve this problem? I am facing the same challenge... *Thanks!

Hello!
My solution was kinda strange. At first, I changed GPIO cable. And I did described procedure for 2-3 times and on the 4th time sticks flashes correctly! Idk what does it mean lol, but I successfully flashed CC2531s ))