Koenkk/Z-Stack-firmware #411: CC2652P As a Router immediately disconnects from coordinator

I-Would-Like-To-Report-A-Bug-Please, 2023-01-20:
Hi,
I've flashed [this CC2652P](https://www.aliexpress.com/item/1005004443820263.html?spm=a2g0o.order_list.order_list_main.5.d60e1802y4ujTP) with the router firmware.  

While I successfully paired it to the SONOFF Zigbee 3.0 it seems that the connection is lost immediately after connecting.  

The link quality shouldn't be an issue. For now, I have both sticks in the same room and while it works I can see lqi around 140.  

I am not sure what is happening. When I try to reflash or push the button on the stick on Zigbee2mqtt I can see:
1) The stick leaving the network
2) Connecting successfully again
3) Interview done
4) Connection dropped for whatever reason ...

I've tried with the:  CC1352P2_CC2652P_launchpad_router_20221102.zip & CC1352P2_CC2652P_launchpad_router_20220125.zip firmware and on both, it behaves the same way.

Is there something that I've missed?

(Also thank you for your amazing work)

Razzer90, 2023-01-20:
I have a similar problem! The dongle runs with the firmware: 6.10.01.01.
As soon as I try to set the transmission rate to 20dbm, the stick drops out! Is this a stable firmware? I find unfortunately no difference between this firmware and the master version.

I-Would-Like-To-Report-A-Bug-Please, 2023-01-21:
I have just tried to hook it as a router to another CC2652 stick(different coordinator, not Sonoff) and pretty much the results are the same.

Razzer90, 2023-01-21:
I was able to solve the problem for myself! A simple restart of z2m and everything has been stable for 12 hours! The router went online after the restart and the transmission rate can be changed at will without the stick going offline! maybe it will help you too!

watercrossing, 2023-01-25:
I own this stick too, but I have flashed it with the coordinator firmware `CC1352P2_CC2652P_launchpad_coordinator_20220219.zip`. It works fine, until it just stops receiving messages.

When no messages have appeared in a while and I do a network scan, the following error message appears:
```
info  2023-01-25 16:33:55: Starting network scan (includeRoutes 'false')
error 2023-01-25 16:34:01: Failed to execute LQI for 'Coordinator'
debug 2023-01-25 16:34:01: Error: SREQ '--> ZDO - mgmtLqiReq - {"dstaddr":0,"startindex":0}' failed with status '(0xc2: NWK_INVALID_REQUEST)' (expected '(0x00: SUCCESS)')
    at Object.func (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

```

Restarting Z2M (not even the pi) fixes the issues for a few hours. After restarting network scans succeed. 

Could this be the same issue?

Smandurlo, 2023-02-07:
I think it is related to my problem: https://github.com/Koenkk/zigbee2mqtt/issues/15912
I cannot change the transmit power and the router disconnects from the zigbee net. If you push the reconfigure button, it will connect back to the zigbee net. It happens only with the last router added, the previous routers work fine.

SK-StYleZ, 2023-02-21:
Hi everybody, i've exactly the same issue.
I'm using two SONOFF Zigbee 3.0 USB Dongle CC2652P + CP2102N, one as coordinator and a second one as router.
I've tried two versions combination of the z-stack fw:
- CC1352P2_CC2652P_launchpad_coordinator_20221226 & CC1352P2_CC2652P_launchpad_router_20220125
- CC1352P2_CC2652P_launchpad_coordinator_20221226 & CC1352P2_CC2652P_launchpad_router_20221102
The router is still disappearing right atfer pairing.
There is one odd thing i've noticed while pairing, the router is announcing it self as "Coordinator", here is the log (newest log entry on top, oldest on bottom):
`
Interview successful: {"friendly_name":"Coordinator","model":"ti.router","vendor":"Custom devices (DiY)","description":"Texas Instruments router","supported":true}
New device joined '0xuuid...' model ti.router
Device 'Coordinator' announced itself
Interview started: Coordinator
`
I am currently running ioBroker with the ZigBee Adapter and planning to try zigbee2mqtt bridge, maybe it's an ioBroker/ZigBee Adapter issue.
Is someone experiencing this issue with zigbee2mqtt?

SK-StYleZ, 2023-03-07:
Hi everybody, I've solved my issues.
I switched to zigbee2mqtt, used the latest firmware for coordinator & router and set transmit power for both to +20dbm.
Everything works now fine.

Smandurlo, 2023-04-07:
Nope

Smandurlo, 2023-05-08:
Nope

Smandurlo, 2023-06-08:
Not sure why this problem is ignored 🙄

ginkel, 2023-06-13:
> I switched to zigbee2mqtt, used the latest firmware for coordinator & router and set transmit power for both to +20dbm. Everything works now fine.

How did you manage to set the Tx power? I can't get my stick to reliably join the network, so setting the Tx power via z2m is quite tricky.

Apart from that, does the coordinator firmware log any debug messages via serial UART that coule be of any help?

SK-StYleZ, 2023-06-16:
> > I switched to zigbee2mqtt, used the latest firmware for coordinator & router and set transmit power for both to +20dbm. Everything works now fine.
> 
> How did you manage to set the Tx power? I can't get my stick to reliably join the network, so setting the Tx power via z2m is quite tricky.
> 
> Apart from that, does the coordinator firmware log any debug messages via serial UART that could be of any help?

I've set the tx power with z2m ui, make sure that coordinator and router are physically close enough.

closed 2023-07-25 by github-actions[bot]