Koenkk/Z-Stack-firmware #416: Share prebuilt files on releases, not inside branches in the git

RubenKelevra, 2023-01-28:
Hey @Koenkk,

currently you're adding the prebuilt binaries in zip archives into branches to the git repo:

* "20221226" (in branch '6.10.01.01')
* "20221102"  (in branch 'develop')
* "20220219" (in branch master)

It would be much easier to find and receive notifications on updates if you could just tag the git revision you're building a specific version off and make a release out of it.

Instead of adding the binaries to the git, you could then add the binaries to the release.

This way we can get a notification from GitHub about the new release.

Thanks!

Koenkk, 2023-01-29:
This is what I already do: https://github.com/Koenkk/Z-Stack-firmware/releases , `20221226` will be released in a few days from now.

ankushg, 2023-02-08:
Nice!

I'm guessing that part of the request is to actually upload the `.zip` file as files on the release, rather than linking to them?

Also: any chance of also creating Github Releases from `develop` versions, and explicitly marking them as pre-release in the Github UI?

closed 2023-03-18 by github-actions[bot]