Koenkk/Z-Stack-firmware #417: Tube's Zigbee PoE (Power Over Ethernet) Serial Coordinator (CC2652P2 variant) 20221226 Firmware update failed

WernerRH21, 2023-01-29:
[image]

tube0013, 2023-01-29:
What are you using to flash?

Is the .hex file extracted?

WernerRH21, 2023-01-29:
Z-Stack-firmware/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip at master * Koenkk/Z-Stack-firmware (github.com)<https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip>
CC1352P2_CC2652P_launchpad_coordinator_20221226.hex

WernerRH21, 2023-01-29:
***@***.***

twinkie0101, 2023-02-07:
I tried to do the update today via bsl-master and receiving the following error after update starts:

`ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'`

I am updating the tubes coordinator (poe version) with the following file: C1352P2_CC2652P_launchpad_coordinator_20221226.hex

tube0013, 2023-02-07:
Please open an issue on my repos for issues flashing TubesZB devises 

https://github.com/tube0013/tube_gateways

closed 2023-03-18 by github-actions[bot]