Koenkk/Z-Stack-firmware #422: After update 20221226 Led indicator is disable, Z-Stack_3.x.0

adamnet01, 2023-02-06:
I have a Z-Stack 3.x.0 Lan coordinator.
On the previous version 20220219, the blue LED was on and when it was pairing, it blinked blue. The second red flashed as the devices communicated with the coordinator.
On the latest version 20221226, only the white LED from the power supply is on.  The other two have no function.  

Is this how it's supposed to be or is there something to adjust.

lux73, 2023-02-07:
which Coordinator exactly?

CC2652RB works as expected - blue LED if is in pairing Mode

adamnet01, 2023-02-08:
I have a device like in the screenshot.  It has three LEDs.  White is power, blue and red.  On the software supplied with the device, it was blue when I was connected to the coordinator and blinked when it was in pairing mode.  Red flashed when devices were communicating. 
The second screenshot is from an earlier version.

[image]
[image]

adamnet01, 2023-02-08:
I searched a bit and had soft from egony with led support.  And the new one from koenkk does not support led.
https://github.com/egony/cc2652p_cc1352p_RF-STAR/tree/main/firmware/coordinator

closed 2023-04-01 by github-actions[bot]

adamnet01, 2024-05-06:
Could someone create a firmware with LED support for the latest 20230507, as in this version from the link?
https://github.com/egony/cc2652p_cc1352p_RF-STAR/tree/main/firmware/coordinator
The version from the link was originally in a device with 3 LEDs.