Koenkk/Z-Stack-firmware #423: Flashing Router Firmware Z-Stack_3.x.0 "CC1352P2_CC2652P_launchpad_router_20221102.zip" fails with "Unknown record type: 3."

rwunsch, 2023-02-08:
I am using "SmartRF Flash Programmer 2" to Flash the Sonoff Zigbee Sticks with the Z-Stack-Router-firmware:
https://notenoughtech.com/home-automation/sonoff-zigbee-3-0-usb-dongle-plus/

When attempting to Flash the just released router-version "20221102" ("CC1352P2_CC2652P_launchpad_router_20221102.hex"), I cannot flash it, and flashing fails with: "Unknown record type: 3."

**Screenshot - CC1352P2_CC2652P_launchpad_router_20221102.hex**
[image]

**Log - CC1352P2_CC2652P_launchpad_router_20221102.hex**
`>Initiate access to target: COM9 using 2-pin cJTAG.
>Reading file: C:/Users/wunsch/OneDrive/Home Automation/Zigbee Sonoff 3.0 USB plus/Firmware/Router/CC1352P2_CC2652P_launchpad_router_20221102.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.`

Flashing with the previous version of the Router works perfectly: CC1352P2_CC2652P_launchpad_router_20210128.hex

**Screenshot  - CC1352P2_CC2652P_launchpad_router_20210128.hex:** 
[image]

**Log - CC1352P2_CC2652P_launchpad_router_20210128.hex:**
>Initiate access to target: COM9 using 2-pin cJTAG.
>Reading file: C:/Users/wunsch/OneDrive/Home Automation/Zigbee Sonoff 3.0 USB plus/Firmware/Router/CC1352P2_CC2652P_launchpad_router_20210128.hex.
>Start flash erase ...
>Erase finished successfully.
>Start flash programming ...
>Programming finished successfully.
>Start flash verify ...
>Skip verification of unassigned page: 24.
>Skip verification of unassigned page: 25.
>Skip verification of unassigned page: 26.
>Skip verification of unassigned page: 27.
>Skip verification of unassigned page: 28.
>Skip verification of unassigned page: 29.
>Skip verification of unassigned page: 30.
>Skip verification of unassigned page: 31.
>Skip verification of unassigned page: 32.
>Skip verification of unassigned page: 33.
>Skip verification of unassigned page: 34.
>Skip verification of unassigned page: 35.
>Skip verification of unassigned page: 36.
>Skip verification of unassigned page: 37.
>Skip verification of unassigned page: 38.
>Skip verification of unassigned page: 39.
>Skip verification of unassigned page: 40.
>Skip verification of unassigned page: 41.
>Skip verification of unassigned page: 42.
>Page: 0 verified OK.
>Page: 1 verified OK.
>Page: 2 verified OK.
>Page: 3 verified OK.
>Page: 4 verified OK.
>Page: 5 verified OK.
>Page: 6 verified OK.
>Page: 7 verified OK.
>Page: 8 verified OK.
>Page: 9 verified OK.
>Page: 10 verified OK.
>Page: 11 verified OK.
>Page: 12 verified OK.
>Page: 13 verified OK.
>Page: 14 verified OK.
>Page: 15 verified OK.
>Page: 16 verified OK.
>Page: 17 verified OK.
>Page: 18 verified OK.
>Page: 19 verified OK.
>Page: 20 verified OK.
>Page: 21 verified OK.
>Page: 22 verified OK.
>Page: 23 verified OK.
>Page: 43 verified OK.
>Verification finished successfully.
>Reset target ...
>Reset of target successful.`

>Initiate access to target: COM9 using 2-pin cJTAG.
>Reading file: C:/Users/wunsch/OneDrive/Home Automation/Zigbee Sonoff 3.0 USB plus/Firmware/Router/CC1352P2_CC2652P_launchpad_router_20221102.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

MicroZot, 2023-02-09:
I also have the same problem with SONOFF Zigbee 3.0 USB dongle-p and Texas flash programmer:
>Initiate access to target: COM8 using 2-pin cJTAG.
>Reading file: C:/Users/AC12/Desktop/AAAAAAAAAAAAAAA TEMP/CC1352P2_CC2652P_launchpad_router_20221102 (1)/CC1352P2_CC2652P_launchpad_router_20221102.hex.
>Unknown record type: 3.
>Reset targets ...
>Reset of target successful.
If I launch the update for coordinator version (CC1352P2_CC2652P_launchpad_coordinator_20221226.hex),
everything works fine...

closed 2023-02-09 by Koenkk

rwunsch, 2023-02-12:
Thanks, @Koenkk !
Sorry for the bother, and that I did not attempt a different flashing tool.
I did not even consider that!!

mom4o, 2023-07-11:
Same situation  is it a solution?

>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: C:/Temp/Sonoff/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

yanirel, 2024-02-13:
same hear

miaszh, 2025-02-14:
any fix?