Koenkk/Z-Stack-firmware #424: Router range decreased

woodrook, 2023-02-10:
Hello, updated my two zigbee router yesterday. two device that were working well doesn't work well anymore since the upgrede. Apperar and disappear and the imput arrive sometimes and sometimes (more) not. Seems to be a definetely range decrease

TomaszDom, 2023-02-11:
I have definitely noticed a severe downgrade in stability going from CC2652RB_coordinator_20220219 to CC2652RB_coordinator_20221226 on my slaesh zigbee coordinator (40 devices). Network became unreliable - lights not turning on, devices not communicating back their status which severely hampered HA automations.

The stroke that broke the camel's back was yesterday when I started getting phantom button presses(!!!) on my MiJia wireless switches (WXKG01LM). I am 100% sure nobody touched them, the logs were clear. None of this ever happened for the 10 or so months I was running the previous firmware. Why, why did I even upgrade if everything was fine before? I downgraded right now and will observe the situation...

TomaszDom, 2023-02-15:
It was not my imagination, the older firmware makes my network rock solid like it used to be. Obviously YMMV.

andi-bar157, 2023-02-17:
> I have definitely noticed a severe downgrade in stability going from CC2652RB_coordinator_20220219 to CC2652RB_coordinator_20221226 on my slaesh zigbee coordinator (40 devices). Network became [...]

To which Version did you downgrade? Is your network fine now?

TomaszDom, 2023-02-17:
CC2652RB_coordinator_20220219 (I was using this before) and yes it is fine now.

Based solely on my personal experience with CC2652RB_coordinator_20221226, it is completely unacceptable as a stable release.

technyon, 2023-03-21:
I'm running 20221226 for several weeks, no issues at all.

TomaszDom, 2023-04-07:
> where can be downloaded older versions of the firmware ?

https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin

closed 2023-05-08 by woodrook