Koenkk/Z-Stack-firmware #442: failed to apply patch when I compile the firmware

Sheng2216, 2023-04-07:
Hi, I am trying to compile the firmware but I got the following errors:

[image]

Sorry if this is a dumb question, I followed the procedures listed in /coordinator/Z-Stack_3.x.0/COMPILE.md, I placed the `firmware.patch` under `workspace_v12` as shown below:
[image]
Then I modify the `Minimal Poll Period (ms)` to 1000 ( without saving), and then change it back to 100 and save the file.
I also checked the versions, I am using `SIMPLELINK-CC13XX-CC26XX-SDK 6.10.01.01`, but for the Code Composer Studio, I am using  `Version: 12.2.0.00009 `

Koenkk, 2023-04-08:
It fails on `syscfg`, did you do step 5 from https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md ?

Sheng2216, 2023-04-10:
> It fails on `syscfg`, did you do step 5 from https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/COMPILE.md ?

"I did follow step 5, and then apply the patch and built the firmware with the command `git apply --reject firmware.patch --ignore-space-change`. The firmware I built works for the CC2652R1, which is great! However, I also want to test the firmware on the CC2652R7. Unfortunately, not only does the firmware I built not work on the R7, but the hex file extracted from 'coordinator/Z-Stack_3.x.0/bin/CC2652R_coordinator_20221226.zip' doesn't work either.

I then built a firmware based on the `znp_LP_CC2652R7_tirtos_ticlang `and flashed it to the R7. At least the LED turned on this time, but I got an error from Z2M, as shown below:

> Zigbee2MQTT:error 2023-04-10 10:28:58: Failed to start zigbee
> Zigbee2MQTT:error 2023-04-10 10:28:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> Zigbee2MQTT:error 2023-04-10 10:28:58: Exiting...
> Zigbee2MQTT:error 2023-04-10 10:28:58: Error: SRSP - SAPI - readConfiguration after 6000ms
>     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
>     at listOnTimeout (node:internal/timers:569:17)
>     at processTimers (node:internal/timers:512:7)
> Using '/app/data' as data directory

My question is, does the `firmware.patch` supports CC2652R7? If not, can I reuse most parts of the patch, as the R1 and R7 are very similar?

mbenfe, 2023-04-10:
Hello
I spent plenty of time trying to fix this issue. I finaly made it work by using the patch system directly from code composer studio.
go to the project root, right click ->Team->apply patch, select the firmware.patch file and then run. it worked perfectly for me and remove issues from git bash. 
I am now facing a different challenge as the generated hex file is a bit bigger than the one provided (410k vs 388K), i will try to find wich option make this change... 
hope it helps

closed 2023-04-13 by Sheng2216