Koenkk/Z-Stack-firmware #454: Z-Stack 3.X for CC2530

alexgeek, 2023-06-08:
Hey, 

I've got a few of [these boards](https://www.aliexpress.com/item/1005003670231117.html). 
I'm not to get any response issuing AT commands using a USB to TTL converter nor wiring it directly to a le potato.

I tried following the instructions from [this guide](https://zigbee.blakadder.com/flashing_ccloader.html) and flashed the [CC2530_DEFAULT_20190608](https://github.com/Jason2866/CCLoader/blob/master/Bin/CC2530_DEFAULT_20190608.zip) firmware from there onto two of them but still nothing comes back.

Is there a 3.X firmware on here that will work with these boards? I'm a bit stuck now. 
Any other advice would be welcome too.

Thanks

lux73, 2023-06-10:
Z-Stack 3.x are nor recommended for these old Coordinators - they are cheap,unreliable & outdatetd

in my Opinion it is not worth to engage with it

pls consider upgrade to a up-to-date CC2652x Chipset - they work really stable

alexgeek, 2023-06-11:
Fair, next time I buy some I'll look at modules with newer chips. 
I'll try some of the 3.0.X firmwares so these don't just turn into e-waste but I'm bit confused why I can flash it but no able to get any UART working. 
I've got TX -> P03, RX -> P02 but no response.

closed 2023-07-26 by github-actions[bot]