Koenkk/Z-Stack-firmware #455: Unknown record type: 3

direx1, 2023-06-14:
Trying to flash CC1352P2_CC2652P_launchpad_router_20221102.hex and get the error Unknown record type: 3. I'm able to flash the coordinator firmware without issue using Flash Programmer 2.

redlingg, 2023-06-15:
same here....(sonoff usb stick)

direx1, 2023-06-15:
> same here....(sonoff usb stick)

I actually found a closed comment on this (several actually after some searching). The issue is with Flash Programmer 2 but a recommendation was to use ZigStar (multitool). The UI is basic and It's quite buggy and took a few attempts to work. Try everything, like insert and write at same time and I did get it flashed eventually.

aj1322001, 2023-06-15:
what are the "correct drivers" and also im trying to use Zigstar now, and i keep getting "Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)' driving me nuts, this is the first time ive ever tried to flash something and its a ball ache!!!

direx1, 2023-06-16:
> what are the "correct drivers" and also im trying to use Zigstar now, and i keep getting "Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)' driving me nuts, this is the first time ive ever tried to flash something and its a ball ache!!!
> 
> ok its done, i think maybe i had the device too close to my laptop and it was causing interference?
> 
> either way its as you said, eventually it worked with ZigStar!

Like I said it's buggy but eventually works

Merril-r, 2023-06-22:
I have the same issue. 

Could not get Zigstar to flash

direx1, 2023-06-22:
> I have the same issue.
> 
> Could not get Zigstar to flash

You need the CP210N driver installed. I found inserting at the exact time of pressing flash eventual worked

Merril-r, 2023-06-22:
How do I get that driver? Windows does not install that one when I install
the driver

azsystem, 2023-06-23:
> How do I get that driver? Windows does not install that one when I install the driver
> […](#)

https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads

azsystem, 2023-06-23:
Same problem here, using the Slae'sh stick.
Tried to flash the old firmware, worked fine (CC2652RB_coordinator_20221226)

Koenkk, 2023-06-23:
This is because of a bug in flash programmer 2, use Uniflash (8.1 should do it)

closed 2023-06-23 by Koenkk

redlingg, 2023-06-24:
Uniflash 8.1.0.4135 does not recognize the stick. Zigstar seems to be ok but HA does not recognize as router. With Zigstar I "can flash" the stick but cannot use it HA.

direx1, 2023-06-24:
> Uniflash 8.1.0.4135 does not recognize the stick. Zigstar seems to be ok but HA does not recognize as router. With Zigstar I "can flash" any fw but cannot use it HA.

Did you previously use this router as a coordinator and migrate to a different coordinator?

redlingg, 2023-06-25:
> > Uniflash 8.1.0.4135 does not recognize the stick. Zigstar seems to be ok but HA does not recognize as router. With Zigstar I "can flash" any fw but cannot use it HA.
> 
> Did you previously use this router as a coordinator and migrate to a different coordinator?

I bought it as used..with the newest router fw flashed. Yes it was used before...

v1k70rk4, 2023-11-02:
It worked for me by pressing the BOOT button continuously during the flashing process, and it just flashed the firmware smoothly. Though, this is strange because this wasn't necessary with the old firmwares.

markusg1234, 2024-02-01:
I couldn't get Uniflash to work.
I Followed these instructions to get it to work with Flash Programmer 2. 

https://github.com/Koenkk/Z-Stack-firmware/issues/397#issuecomment-1510412650

.\srec_cat.exe "CC1352P2_CC2652P_launchpad_coordinator_20230507.hex" -intel -o CC1352P2_CC2652P_launchpad_coordinator_20230507.bin -binary