Koenkk/Z-Stack-firmware #459: Corrupted archive of the firmware

alex-savin, 2023-06-23:
Hey Folks,

I was trying to download a firmware, but all of the time I'm getting the error the archive is corrupted.
https://github.com/Koenkk/Z-Stack-firmware/blob/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip
Could you please check it 

Thank you in advance!

Koenkk, 2023-06-24:
Works fine here.

kam821, 2023-06-27:
```
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/blob/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip

Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20230507.zip'

sonoff

Opening port /dev/ttyUSB0, baud 500000

Reading data from CC1352P2_CC2652P_launchpad_coordinator_20230507.zip

Cannot auto-detect firmware filetype: Assuming .bin

Connecting to target...

CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8

Primary IEEE Address: 00:12:4B:00:24:C2:A1:45

    Performing mass erase

Erasing all main bank flash sectors

    Erase done

Writing 146881 bytes starting at address 0x00000000

ERROR: Invalid data size: 146881. Size must be a multiple of 4.
```

andregoncalvespires, 2023-07-01:
>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: C:/Dongle/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

rodrigogbs, 2023-07-01:
> Initiate access to target: COM3 using 2-pin cJTAG.
> Reading file: C:/Dongle/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
> Unknown record type: 3.
> Reset target ...
> Reset of target successful.

Same problem here…

gargamelonly, 2023-07-02:
Failing also updating (CC2652R_coordinator_20230507) with hex file on zzh! dongle and TI Flasher.
Previous firmware hex pass ok (CC2652R_coordinator_20221226).

Koenkk, 2023-07-02:
https://github.com/Koenkk/Z-Stack-firmware/issues/455

closed 2023-07-02 by Koenkk