Koenkk/Z-Stack-firmware #462: What is the current procedure for flashing the CC1352P2_CC2652P_launchpad_coordinator_20230507.zip firmware?

andregoncalvespires, 2023-07-01:
What is the current procedure for flashing the CC1352P2_CC2652P_launchpad_coordinator_20230507.zip firmware? Using bootloader mode via python with "uartlog.py" and "Smart RF Flash Programmer2" is not working. Returning the error below:

Initiate access to target: COM3 using 2-pin cJTAG.
Reading file: C:/Dongle/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
Unknown record type: 3.
Reset target...
Reset of target successful.

I believe that this program is not capable of performing this update. Which procedure to use?

tube0013, 2023-07-02:
see https://github.com/Koenkk/Z-Stack-firmware/issues/455

AbeltjeNL, 2023-07-02:
I found this and worked for me:

https://github.com/Koenkk/Z-Stack-firmware/issues/397#issuecomment-1510412650

ahd71, 2023-07-02:
I had the same issue and instead tried with https://github.com/xyzroe/ZigStarGW-MT/releases/ to flash my stick which worked great.

KoKolaj, 2023-07-02:
I used this tutorial - https://smartlight.me/diy-smart-home/flashing_slzb-02

closed 2023-07-03 by andregoncalvespires