Koenkk/Z-Stack-firmware #466: CC2652P won't pair after firmware update

blacknell, 2023-07-17:
I've just updated a brand new Sonoff Zigbee 3.0 USB Dongle Plus to [CC1352P2_CC2652P_launchpad_router_20221102](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20221102.zip) and that worked fine. It immediately paired with the coordinator and all is good.

So I moved to my older one (about a year) and although the flashing went fine, it won't pair. It was paired just fine and I needed to update it.

Looking at the flashing output I see this difference

Older dongle
```
Opening port /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_865e08dede12ec119b8321c7bd930c07-if00-port0, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_router_20221102.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:24:C7:C6:89
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x9877ee56)
```

New dongle
```
Opening port /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_router_20221102.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.1 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:29:E8:4B:54
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x9877ee56)
```

The only difference seems to be that rev number (?) `CC1350 PG2.0` is older.

Any ideas?

blacknell, 2023-07-17:
I've also since tried deleting the router from the network (forced but allow rejoin) but there is still no attempt to pair

blacknell, 2023-07-19:
It turns out you have to open the case and click RESET after powering up so it starts to re-pair.

closed 2023-07-19 by blacknell

tbgoose, 2023-07-24:
I have two spare sonoff CC2652P Sonoff ZIgbee Plus sticks that I would like to use as routers.  I am in the same situation but even with multiple times trying to pair after pressing the reset OR boot button nothing comes up in zigbee2mqtt (in devices or logs).

I have tried flashing CC1352P2_CC2652P_launchpad_router_20220125.zip with https://github.com/JelmerT/cc2538-bsl, https://github.com/git-developer/ti-cc-tool and ZigStar. All indicate that the process has completed successfully, but I can't get my two spare sticks to do anything in zigbee2mqtt.

Is there someway I can see what is actually happening? Perhaps the sticks are both kaput. One was previously working as a router, but I had never updated it so I wanted to do that. The other I couldn't get to pair after updating in April 2022, I guess with the previous router firmware version.

blacknell, 2023-07-24:
I found I had to repair the router after flashing. You need to unscrew the screws at one end. Take out the Board attach aerial. Power up. Then press the tiny reset button for a few seconds and you will see it join. Then power off and reassemble.

blacknell, 2023-07-24:
And 20221102 version

tbgoose, 2023-07-24:
I have tried pressing and holding both the keys, hasn't helped. How can I get my hands on the 20221102?