Koenkk/Z-Stack-firmware #467: Coordinator 20230507 won't pair with Hue wall switch. Downgrade to 20221226 worked.

alexsporn, 2023-07-18:
I updated to the latest 20230507 coordinator firmware for my SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P and everything seemed to work fine.
I then tried to pair some new Hue wall switches I had around and they would not pair. Multiple restarts did not help.
I downgraded again to 20221226 and the Hue wall switch immediately paired on the first attempt.

Cossey, 2023-07-25:
Just wanted to note that I also upgraded to this firmware - things seem fine, however I lost connection to my Phillips Hue 324131092621 Dimmer Switch. When I tried to re-pair to the coordinator, it didn't automatically Bind the Endpoint and pressing buttons did not show the light on the dimmer switch. When I manually added the bindings to the switch in z2m, the light on the dimmer switch flashes, but it appears the action shows nothing on the z2m UI.

Wickedy123, 2023-07-30:
I have three Hue outdoor PIR sensors which go offline every week or so. I fix them via rebooting the pi completely, restarting z2m didn't seem to work. I have the same issue with my Salus sr600 relay. This is new behaviour since 20230507. (However, all my popp trvs are behaving, so in a better place overall).

closed 2023-09-06 by github-actions[bot]