Koenkk/Z-Stack-firmware #468: Programming new Version failed with TI Flashpr.2

Ledikie, 2023-07-30:
Hello,
I'm trying to program the new version "CC1352P2_CC2652P_launchpad_coordinator_20230507.hex" to RFStar module.
But it failed.
When I program the last version "CC1352P2_CC2652P_launchpad_coordinator_20221226.hex" it runs without a problem.
I tried it several time without success. My programmer Texas Smartflash2 with XDS110.

[image]
[image]

lux73, 2023-07-30:
try flashing with [python script](https://github.com/JelmerT/cc2538-bsl) as mentioned here: https://www.zigbee2mqtt.io/guide/adapters/#flashing-cc1352-cc2652-cc2538-based-adapters

works for me

KoKolaj, 2023-07-30:
Works for me - https://smartlight.me/diy-smart-home/flashing_slzb-02

Ledikie, 2023-07-30:
I have only the pure board (RFSTAR RF-BM-2652P2 SMA Ant.). The board is connected with 4 wires direct to my Raspi4 for operation. 2 for power supply and 2 for RX/TX.

To do the programming, I soldered 3 additional wires to the board, and connect it to the XDS110.

What I saw is the difference between the size of the 2 files:
CC1352P2_CC2652P_launchpad_coordinator_20230507.hex    498 KB new one
CC1352P2_CC2652P_launchpad_coordinator_20221226.hex    401 KB older one
Maybe this is the reason why I can’t program the new file.

ahd71, 2023-07-30:
Several threads/issues about the same. Try another tool to flash with (I had the same)

Ledikie, 2023-08-01:
Find a solution to program it with TI Flashprogrammer2.
Followed the instructions by this link: https://github.com/Koenkk/Z-Stack-firmware/issues/397#issuecomment-1510412650
Changed the file from ....hex into ....bin

kgiedrius, 2023-08-10:
I got same error with 2023 firmaware, but 202212 - worked fine.
Smth wrong is with .hex file

Koenkk, 2023-08-10:
The TI flash programmer v2 is broken, it doesn't support support the latest `.hex` files their SDK generate. There is nothing I can fix from the firmware side here. (use TI UniFlash/cc2538-bsl as an alternative).

closed 2023-08-10 by Koenkk

david-milward-r3, 2023-08-10:
It worked fine for me using hex2bin.

kgiedrius, 2023-08-11:
TI flash program is also broken on newest windows version.

kgiedrius, 2023-08-11:
@Koenkk  Maybe you can include .bin version of .hex file so it can be flashed with TI 2 programmer software

lux73, 2023-09-01:
come on Guys - just use the mentioned cc2538-bsl.py Script. It works like a charm