Koenkk/Z-Stack-firmware #481: Unknown record type: 3 with 20230507

smarteshome2020, 2023-10-29:
Hi,

i try to flash my Sonoff Stick from 20220219 to 20230507 and it fails.
Flash from 20220219 -> 202331226 works

Du you have a hint for me? 

>Reset of target successful.
>Initiate access to target: COM6 using 2-pin cJTAG.
>Reading file: E:/Download/CC1352P2_CC2652P_launchpad_coordinator_20230507/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.
>Initiate access to target: COM6 using 2-pin cJTAG.
>Reading file: E:/Download/CC1352P2_CC2652P_launchpad_coordinator_20221226/CC1352P2_CC2652P_launchpad_coordinator_20221226.hex.
>Start flash erase ...

Koenkk, 2023-10-29:
Duplicate of https://github.com/Koenkk/Z-Stack-firmware/issues/455

closed 2023-10-29 by Koenkk

smarteshome2020, 2023-10-29:
thanks and sorry fr the duplicate :-)