Koenkk/Z-Stack-firmware #482: Device is leaving network after just putting it pairing mode

chojnicki, 2023-11-01:
I recently switched adapter from Conbee 2 to Sonoff Dongle-P. I have wall switches (TS0001 and TS0002), but I tested same behavior on other devices like Ikea Tradfri bulbs. So when device is already paired, and I start pairing mode again on it, it just leaves network even that it was not paired to new gateway/coordinator. It makes no sense, it should keep current network if new one was not found. Sometimes I get message in Z2M "Device 'xxxxxx' left the network" and it just disappears from dashboard, but often there is no info in Z2M and it just stops working - not sure why different behaviors. With bulbs or most devices it's not really a big issue, but with switches it is easy to put them in that pairing mode by accident, because it requires just 5 clicks to do that, which often my small kids do...

It is easy to blame device - but most important thing is, like I said it was NOT such issue with previous adapter. Maybe conbee did not "allowed" to leave network by device itself, and z-stack does, not sure. Previously in case of my switch, it was unresponsive for 1 minute because it was looking for new coordinator, and after that time everything was back to normal. But now I have to repair them again manually every time.

So why device is leaving me, just by putting it in pairing mode? Can I somehow disallow leaving network, or is it a bug?

I'm using firmware 20221226. I know it's not latest one, but it is considered as stable, and I did not found any fix related to my issue in changelog.

chojnicki, 2023-12-08:
Still a issue

jonzyk, 2024-01-05:
I switched from an Elelabs ELU013 (ezsp) to a SONOFF Dongle-P. Since I have some strange behavior with my switch, I wanted to try a recommended adapter.
I faced the same issue with 20230507
Devices already paired with my ezsp are stuck into a pairing loop. I tried several resets of the device with no effect.
New devices are paired OK.
I have seen this thread before : https://github.com/Koenkk/Z-Stack-firmware/discussions/474
So I downgraded to 20221226, and was able to pair with only 1 reset of the device.

chojnicki, 2024-02-06:
bump to unstale

chojnicki, 2024-07-02:
@Koenkk could you reopen this please? It's not completed :( I hate this GH bot :D

chojnicki, 2024-08-02:
Unstale

Koenkk, 2024-08-03:
> Can I somehow disallow leaving network, or is it a bug?

No this is not possible, to figure out what is happening I would need to have a sniff of this: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

closed 2024-09-10 by github-actions[bot]